motioneye-project/motioneyeos #1402: View the interface from the pi with the OS?

smallzZz8, 2018-02-22:
Hello I would like to install this on my Rpi but I would also like to dedicate that Rpi to be able to view the interface. What commands can I run to launch the interface from the same Rpi that it is installed on?

maglub, 2018-02-23:
In motioneyeos you cannot do this. It is ment to be "headless", and you access it through a web-browser.

My suggested solution:

* Buy a second RPI and install a full Raspbian on in, and use that to show the web GUI

ccrisan, 2018-02-24:
You can't and you won't be able to do that on one PI, without serious performance penalties. As @maglub said, you'd have to do that from another computer - even if it's another PI.

closed 2018-02-24 by ccrisan