motioneye-project/motioneyeos #1406: Night Vision Not Working with NOIR + IR  LEDS

j0nr, 2018-02-24:
I have a Pi Zero W with Raspbian + Motioneyeos working great in daytime. I have the NOIR Camera with twin IR LEDS. The LEDS are working. 

When lights go out the image is black... Not sure where to go from here.

Thanks in advance

j0nr, 2018-02-25:
I am wondering if I have a defective camera... It doesn't seem to work at all in low light... And I'm talking just dim... See photo attached... Shows camera looking into a cupboard. Just as you see it in the photo, not shutting cupboard door, the camera is just looking into the cupboard and the image goes completely black. 

[image]

Works fine in good daylight

j0nr, 2018-02-25:
This is the camera I bought :Raspberry Pi Zero W Camera Module Night Vision with 3.6mm Adjustable Focal Length 2pcs IR Sensor LED Light Raspberry Pi Zero Adjustable Camera https://www.amazon.co.uk/dp/B0748GQ32H/ref=cm_sw_r_cp_apa_nBPKAbKRMHWYF

And the power supply is up to 3A

mb39924, 2018-02-25:
Some thoughts.. Dit you connect the leds yourself? Do they have a red glow? I think these have two small adjustment screws to set the sensitivity for the leds.

j0nr, 2018-02-25:
Hi,

Yes they have a red glow. If I point the camera towards a reflective surface the camera can see the IR light source... Maybe they are not bright enough? What do the adjustment screws adjust? 

Thanks for the reply

mb39924, 2018-02-25:
Are the colors of the items in your cupboard correct? If so, your camera has an infrared filter and your ir leds won't work. Point the camera outside. Without the leds working the image should show pink and red colors where there are none.

Dewey3, 2018-02-25:
Go into "Expert Settings", and if you have "Enable CSI Camera LED" turned on, turn it off.

j0nr, 2018-02-26:
It's definitely a NOIR camera, without a doubt.

What does "Enable CSI Camera LED" control?

jasaw, 2018-02-26:
@j0nr You shouldn't need to disable the Pi Camera LED, which is the CSI Camera LED option, unless you have the NoIR with built in IR-cut filter. The IR-cut filter is controlled by turning on/off the Pi Camera LED.

j0nr, 2018-02-27:
So I definitely have the NOIR Camera... Am I turning CSI Camera LED on or off? How do I do this? I am using Motioneye sorry, not motioeyeOS

j0nr, 2018-02-27:
Ah... Twiddling the tiny adjustment screws on the LED boards seems to have done it... Tweaked so the LEDs (the main IR ones) go off in light and come on in dark... I think before I tweaked that something was switching in the camera and just blanking the display...

We'll see.

sparkeh, 2018-02-28:
I had a similar issue and same camera module and the fix for me was to edit `config.txt` and add in the following line:

```
disable_camera_led=1
```

AlexFielder, 2018-09-26:
> Ah... Twiddling the tiny adjustment screws on the LED boards seems to have done it... Tweaked so the LEDs (the main IR ones) go off in light and come on in dark... I think before I tweaked that something was switching in the camera and just blanking the display...
> 
> We'll see.

I'm 99% certain I have the same camera/LED combo that you do; which way did you need to turn the screws to get the LEDs to activate correctly?

alpotapov, 2019-06-13:
Here is something that I found. Apparently the camera has GPIO that allows to toggle between normal/night mode: https://www.waveshare.com/wiki/RPi_IR-CUT_Camera

jasaw, 2019-06-18:
@alpotapov The IR-cut filter is controlled by turning on/off the Pi Camera LED. Alternatively, solder a wire from the camera GPIO pad to a Pi GPIO pin, and control the IR-cut filter from the GPIO.

STUKguy, 2020-05-28:
Has anyone fixed this issue? i have the same type of camera but my IR stays on now matter what screws the twist or config line i add?

jasaw, 2020-05-29:
@STUKguy which raspberry pi model are you using?

STUKguy, 2020-05-29:
@jasaw I am using a pi zero w

STUKguy, 2020-05-29:
@jasaw  basically i brought the below
https://www.ebay.co.uk/itm/5-Megapixel-Infrared-Night-Vision-IR-CUT-Camera-For-Raspberry-Pi-4B-3B-Zero-NEW/293554729746?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

Connected it and when i power on the PI Zero W i get this
[image]

And this is what i get when i log in to motion eye os (latest build) bare in mind it is the middle of the day and room i bright
[image]

This is with CSI Camera led off

If i adjust the small screw next to IR thing on the side
[image]

so they are both off 
[image]

i still get this when i log into motion eye
[image]

If i enable CSI and reboot the camera i still get the same thing but when i face the camera towards my PC monitor the colour is fine but when i face the camera around my room the colour is all purple-ish
[image]
[image]

and finnaly if i turn the sreaws again nothing changes.

I have played about ith the colour setting on Motion eye but this does not and like i said when i point it towards my screen all the colours are correct.

Any ideas, sorry about the really long post

jasaw, 2020-05-29:
@STUKguy 
There are a few variants of this camera. 2 major variants are:
- Manual controllable IR-cut filter, controlled via camera LED. I highly suspect you have this one.
- Automatic IR-cut filter. This has a light sensor on the camera PCB.

You are getting red image because your IR-cut filter (inside the camera) is not enabled, and when there's sun light, the infra-red causes red image. First, you need to turn off camera LED from motionEyeOS, then use software to control when to turn on/off the IR-cut filter by turning on/off camera LED.

There's more discussion on IR-cut filter control here: https://github.com/ccrisan/motioneyeos/issues/1013

jasaw, 2020-05-29:
I'm also using this camera with manual IR-cut filter, and I've built a simple [light-dependent resistor](https://www.ebay.com.au/itm/20PCS-Photoresistor-GL5528-LDR-Photo-Resistors-Light-Dependent-WS/222122787897) (LDR or CdS) with [small IR-cut filter](https://www.ebay.com.au/itm/10pcs-Optical-UV-IR-Cut-Light-Filter-Blocking-For-Camera-Astronomy-Lens-8mm-Slim/282653207704) to detect light, and some IR-LEDs for illumination. I've also written a simple [LDR reader](https://github.com/jasaw/ldr-reader) program that reads the LDR and sets GPIOs to high/low and runs commands. It has been working well for me.

STUKguy, 2020-05-29:
@jasaw when you say the turn off the camera LED from motion eye do you mean CSI camera Led
[image]

basically what im trying to create is a outdoor waterproof camra, which has night vision during the night and normal vision during the day i have everything else ready to go but this current issue i am having is slowing me down.

badjano, 2021-11-29:
> @jasaw basically i brought the below [...]

facing the same issues, I have up, I think those lights are for night vision and it won´t turn on and off automatically, I know this because when I film myself really close my eyes turn bright ( my eyes are brown )

jasaw, 2021-11-29:
@badjano I think the IR LEDs and IR-cut filter are controlled separately on these cameras. Either way, there are 2 variants of this camera on the market.
1. IR-LED and IR-cut filter both controlled by hardware. Nothing needs to be done on the software side.
2. IR-LED controlled by hardware (so nothing can be done via software), but IR-cut filter is software controlled, and turned on/off via the camera LED. This requires the CSI camera LED to be disabled via motionEye configuration, and have a separate software control it. There's another variant under this category that comes with a jumper wire that controls the IR-cut filter.

**IR-LED**
How do you know when it is turned on/off? When it's on, it actually looks a little red, kind of like a dim red LED. This has nothing to do with the purple color that you see from the video stream. You might be able to tweak the threshold of when the IR-LEDs turn on/off and this should work independently from the IR-cut filter.

**IR-cut filter**
Why is the video stream purple? It's purple because the camera is exposed to infrared light (from sunlight mostly), which means the IR-cut is not enabled. This allows the camera to see at night (with the help of IR-LED). When the camera (with IR-cut filter disabled) is facing away from the window (meaning less infrared light getting into the camera lens), the video stream is a lot less purple.
After turning **off** the CSI camera LED from motionEye config, you could try setting the CSI camera LED GPIO state high or low to see whether it has any effect on the purple video stream. See here for more discussion about setting CSI camera LED state: https://github.com/ccrisan/motioneyeos/issues/1013