motioneye-project/motioneyeos #1409: synchronize from motioneye to NAS server

AlexNeedsAndProvidesHelp, 2018-02-27:
Hello,

how can i send movies from motioneye (/data/output/Camera1) to the hard disk i have connected to my router ?

I tried with ftp but my router only accept AFP protocol. The problem is that i cannot install package in motioneye in order to make my raspberry able to send data with afp protocol.

Does anybody have a suggestion ?

tuppes, 2018-02-27:
Hi,
are you talking about **motioneye** or **motioneyeos**?

If it is for motioneye you're wrong here.
If iit is for motioneyeos you will find an answer in jasaw's comment of #1258.

Best wishes!

AlexNeedsAndProvidesHelp, 2018-02-28:
hi
thx for your fast answer.

I have motioneyeOS on my raspberry. I wanted to avoid to install motioneye on raspbian but evidently i have no other options. 

so i tryed it and i got many errors and warnings:

> Err:2 http://mirrordirector.raspbian.org/raspbian stretch/main armhf libavdevice      57 armhf 7:3.2.9-1~deb9u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]

>Err:3 http://mirrordirector.raspbian.org/raspbian stretch/main armhf ffmpeg armh      f 7:3.2.9-1~deb9u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]

>E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/f/ffmpe      g/libavdevice57_3.2.9-1~deb9u1_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:100      0:75:0:3 80]

>E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/f/ffmpe      g/ffmpeg_3.2.9-1~deb9u1_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:      3 80]

>E: Unable to fetch some archives, maybe run apt-get update or try with --fix-mis      sing?

>Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found

and so on.

I am cursed 
but i think i wont get any answer here since it is only the motioneyeOS community.

thank you tuppes 4 ur answer

bye

jasaw, 2018-03-03:
Use rsync. It's more reliable than FTP.

jasaw, 2018-03-03:
@AlexNeedsAndProvidesHelp If your NAS supports `rsync`, you could try something like this on MotionEyeOS:

```if [ -z `pidof rsync` ]; then timeout -t 1800 nice -n 19 rsync -av --delete --bwlimit=800 --exclude='.keep' -e "nice -n 19 ssh -i /data/etc/ssh_host_rsa_key -o UserKnownHostsFile=/data/etc/known_hosts" /home/ftp/sdcard/ $NAS_USERNAME@$NAS_HOSTNAME:$MEDIA_UPLOAD_PATH/ ; fi```

This command does a few things:
- only run rsync if not already running
- limit rsync session to 30 minutes, in case of a stalled transfer
- runs with the lowest CPU priority
- limits transfer rate to 800 kbytes/s
- connects via ssh with key login so no password required
- uploads to $NAS_HOSTNAME as user $NAS_USERNAME
- finally, uploads /home/ftp/sdcard to $MEDIA_UPLOAD_PATH

OldHolborn01, 2018-03-06:
Hi, thank you for the Idea!
sorry, I´m not very firm in Linux...

1st. is it right to put the command to FileStorage: "Run a Command" in Motioneye?

2nd, I would like to synch my Files to a Network-Server (NTFS/Samba, Windows) like
//ip/path/ + userename + password

do you can explaine the the correct syntax for the command?

Thank you for help

jasaw, 2018-03-06:
@OldHolborn01 I don't think Samba is suitable for automatic backup. If you still want to run rsync on Windows, but you could install [cwrsync](https://itefix.net/cwrsync) on Windows.

AlexNeedsAndProvidesHelp, 2018-03-06:
finally i am happy to have installed motioneye. Installation guide works fine [(link)](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) and i even succed to transfert data by mounting the storage device in raspbian . This tuto helped me a lot to mounting : https://rasspberrypi.wordpress.com/2012/09/04/mounting-and-automounting-windows-shares-on-raspberry-pi/ 

than i just have to  push my movie folder in the storage device with a cp command. Nothing to install, no protocol are needed. 

 @jasaw : rsync is very bad because it requires a password. you cannot use it in the end command of motioneye.

for newbies, try to understand what mounting is.

jasaw, 2018-03-06:
@AlexNeedsAndProvidesHelp I'm glad that you've got your set up sorted out.

Just to clarify if anyone is interested, `rsync` does **not** require password if you use SSH key to log in. See `ssh -i /data/etc/ssh_host_rsa_key` part of the command.

AlexNeedsAndProvidesHelp, 2018-03-07:
with rsync you can generate a ksa key with no password or you can download sshpass and write your password in a folder. Both are not secure and a bit tricky.

AlexNeedsAndProvidesHelp, 2018-03-07:
okay. I have tested rsync and you cannot use it because it requires sudo command and motioneye doesnt allow the sudo command.

jasaw, 2018-03-07:
@AlexNeedsAndProvidesHelp MotionEye runs as root on MotionEyeOS, so not an issue. `rsync` works on MotionEyeOS. I'm using it on my own set up because I find it's the most reliable way of transferring big media files.

AlexNeedsAndProvidesHelp, 2018-03-08:
> MotionEye runs as root on MotionEyeOS

yes but i run motioneye and not motioneyeOS. So i cannot :(

closed 2018-03-10 by ccrisan