motioneye-project/motioneyeos #1410: PI 3 w/multi camera adapter board?

eight08, 2018-02-27:
Is it possible to use the multi camera adapter board for pi with motioneyeos?   Does anyone have experience with these boards?

jasaw, 2018-02-28:
@eight08 Are you talking about Raspberry Pi compute module with dual CSI ports?

eight08, 2018-02-28:
Sorry I should I have been more specific.  I'm referring to the ArduCAM 4 cam adapter board for pi
http://www.arducam.com/multi-camera-adapter-module-raspberry-pi/

jasaw, 2018-02-28:
It won't work. From the product page, it says:
> For proper operation, only one camera should be enabled at a time.

I looked into this adapter board before. It works by switching the CSI signals to whichever camera that is selected, i.e. cameras can't work simultaneously. If you want simultaneous operation of multi-camera, you need the Compute Module, but I don't know if motion supports it.

eight08, 2018-02-28:
Ah i see. Thank you for clarifying this for me!
I take it you mean a Compute Module and I/O board?

eight08, 2018-02-28:
Ah thank you for clarifying this for me. Yeah the compute module would get expensive pretty quickly for what i'm trying to accomplish i'm afraid... but it looks pretty cool! I need, ideally, to have four IR cameras in close proximity that I can log into remotely.  I suppose I can use USB cameras on the Pi 3 instead...?

jasaw, 2018-02-28:
Multiple cameras on a Pi3 might work, but keep in mind the Pi limitations, like USB bandwidth, memory bandwidth, CPU. It comes down to your video resolution and frame rate.
Maybe someone who has actually tried this set up can share their experience?

giac0m0, 2018-03-02:
Maybe you'd be better using a PiZero for each cam as network cameras and seup a powerful server on the network to combine the streams into one user interface on a single IP.

UrbanSwati, 2018-03-08:
@giac0m0 Interesting, have you tried it?

giac0m0, 2018-03-08:
https://github.com/ccrisan/motioneyeos/wiki/Usage-Scenarios

I've got a Pi0w CSI Camera Fast Network Camera to a Pi3 . The Pi3 has a CSI camera module, with both cameras appearing on the user interface as a hub.  

If I add more cameras, I will either invest in a more powerful hub or run the MotionEye software on a networked server.