motioneye-project/motioneyeos #1411: downloaded video won't play properly after upgrade to 20171008

atvinnys, 2018-02-28:
Hi all

I can play the video recordings via the samba server just fine. (win10 terminal, using VLC)
but if I transfer the mp4 either via a simple copy to my desktop from an explorer or via the download option of the web GUI, and then play the video, it will not play properly: it shows the first image only and rolls thru time but the video is stuck on that first image.

I went back in time and I have not this problem with the previous version I was using (maybe 2 versions back). 
not quite sure what to make of this....
any idea?
thanks
vincent

jasaw, 2018-02-28:
What did you use to play the recorded video? Try another player (not VLC).

mb39924, 2018-02-28:
I had this problem also  with previous versions using vlc. For motioneyeos mp4 downloads I use windows mediaplayer.

atvinnys, 2018-02-28:
I also remembered I upgraded VLC around the same time ..so I loaded up MPC and it works there.
so VLC is the issue here.

thanks for the input!

closed 2018-02-28 by atvinnys