motioneye-project/motioneyeos #1416: Wifi preconfiguration problem

berniemci, 2018-03-03:
Hi, I followed the guide for Windows wifi preconfiguration to set up wpa_supplicant.conf. When I start the initial boot without an ethernet connection, it gets stuck on configuring wired network and goes into a loop.

Are the instructions up to date. I have a tried a number of times already without success.

jasaw, 2018-03-03:
Presumably you followed this wifi pre-configuration guide: https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration
Have you read the wifi troubleshooting page? https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems

Do you mind posting your wpa_supplicant.conf file with SSID and psk masked out?
What hardware do you have? Is your wifi adapter supported? Some more details on your hardware set up might help.
Which version of MotionEyeOS are you running?

berniemci, 2018-03-03:
Thanks for your response, Please close the issue. When using Notepad the file was saved as a text file in error. The instructions are fine.

closed 2018-03-03 by berniemci