motioneye-project/motioneyeos #1418: Using access point mode

lwild12, 2018-03-03:
When trying to boot my raspberry Pi zero W after being configured following https://github.com/ccrisan/motioneyeos/wiki/WiFi-Access-Point
my pi will boot, get to "configuring wired network" and then will output " network: no network connection available, rebooting" before shutting down. It does this on an endless cycle

ccrisan, 2018-03-03:
@lwild12 do you actually have a network connection? If you don't have one, set [`os_networkless`](https://github.com/ccrisan/thingos/wiki/os.conf#os_networkless) to `true`.

lwild12, 2018-03-04:
@ccrisan I now have that set to true although the board is now booting it isn't broadcasting the wireless network. I have attached my os.conf, hostapd.conf and dnsmasq.conf as text files
[hostapd.txt](https://github.com/ccrisan/motioneyeos/files/1778368/hostapd.txt)
[os.txt](https://github.com/ccrisan/motioneyeos/files/1778369/os.txt)
[dnsmasq.txt](https://github.com/ccrisan/motioneyeos/files/1778370/dnsmasq.txt)

ccrisan, 2018-03-04:
Here are a few remarks:

1. Your `os.txt` file (assuming it's actually taken from `/data/etc/os.conf`) contains spaces around the equal sign. Please remove them.
2. Your `dnsmasq.txt` file (again assuming it's taken from `/data/etc/dnsmasq.conf`) has a space in front of each line. Please remove them.
3. After fixing these issues, please run the system again and, if it still doesn't work, attach your log files (notably `boot.log`, `dmesg.log`, `hostapd.log` and `messages`).

laurent1981be, 2019-07-24:
Hi ccrisan, 
Thanks for creating MotionEye, its funny to test and try to run.
But...
I've a Pi Zero W, as Iwild12 and i can't run the AP.
I tried many times and find some errors ont my .conf files. Spaces principaly
Modified now but i already have an error with the "wire network"...
Here are my files, i hope you can help me to make my AP run on the Pi Zero W;
Thanks in advance,
[os.txt](https://github.com/ccrisan/motioneyeos/files/3428024/os.txt) put in data/etc at first boot
[dnsmasq.txt](https://github.com/ccrisan/motioneyeos/files/3428026/dnsmasq.txt) put in /boot  at first boot
[hostapd.txt](https://github.com/ccrisan/motioneyeos/files/3428027/hostapd.txt) put in /boot  at first boot
Here are the log files

[boot.log](https://github.com/ccrisan/motioneyeos/files/3428036/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3428037/dmesg.log)
[hostapd.log](https://github.com/ccrisan/motioneyeos/files/3428038/hostapd.log)
[messages.txt](https://github.com/ccrisan/motioneyeos/files/3428040/messages.txt)

sgomon, 2020-12-05:
I'm having the same problem.  Do you just put the os.conf file in the root directory of the SD card with the dnsmasq.conf and hostapd.conf files before booting the first time?  If setting up a static ip, do you put the static_ip.conf file in the same place as the other files?  Is there a sample os.conf and static_ip.conf file anywhere that I can use as a starting point?

starbasessd, 2020-12-05:
Not in root, you need to put all of them in /boot before first boot, including wpa_supplicant.conf, if used.
If you use os.conf, you effectively disable dhcp and MUST use a static IP.
If you use a static_ip.conf file, you would be disabling getting a dhcp address if your 
wpa_supplicant.conf file is connecting to another wifi access point.
Attached is a zip file of the config files to set up an AP setup, and instructions.
If you edit (and you should) use a Linux friendly editor like NotePad++, not windows default Notepad.
 
 
[ConfigFiles(20200606-or-newer).zip](https://github.com/ccrisan/motioneyeos/files/5646272/ConfigFiles.20200606-or-newer.zip)

starbasessd, 2020-12-05:
If you are using an older image, (pre-20200606)  use these files instead

[ConfigFiles.zip](https://github.com/ccrisan/motioneyeos/files/5646280/ConfigFiles.zip)

sgomon, 2020-12-05:
Great.  I'll give this a shot.  I did setup another Raspberry PI webcam where I copied the wpa_supplicant.conf to the root folder before booting and that did seem to work.  It was able to log on to my wireless network.  The one I am working on now is going on a robotic tank to allow you to drive using a VR headset, so it needs to be it's own network.

starbasessd, 2020-12-05:
Cool.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sgomon, 2020-12-05:
Thanks a lot.  It worked.  Kind of.  Once in a while I can connect to the Raspberry PI Zero W with my phone.  Most of the time the PI shows up on my phone WI-FI connections, but disappears when I try to connect.  I believe it is a PI issue, not this software.  I'll try it on a PI 3 or better to see if it works and maybe go that direction.  At least I have everything I need to configure it and get it working as an access point camera server.

sgomon, 2020-12-05:
I think I figured it out.  I was using a night vision camera with illuminators.  I don't think the PI Zero could power them,  I took the illuminators off and it looks like it is working now.

starbasessd, 2020-12-05:
Good to hear. Enjoy.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2020-12-05 by starbasessd

danger355, 2024-02-10:
@starbasessd I'm using the four config files you provided, and I've gotten as close as ever to completing my goal, so thanks for these.

my problem is, I'm still using an IP/PoE cam (I will be using multiple in the end), but it seems the camera isn't accessible. this is confirmed by a) locking the ip of the camera within the dhcp range in your dnsmaq.conf, and b) using a second camera with dhcp still enabled then rebooting the pi to make sure everything should be accessible.

it seems that `OS_NETWORKLESS="true"` kills all _wired_ network functions, since I can get to the MotionEyeOS dashboard using the access point set up in hostapd.conf ... is this correct?

turning `OS_NETWORKLESS="false"` causes the no network panic reboot loop.

is there a way to disable looking for the 'internet' while keeping wired networking so that I can use PoE cameras?

starbasessd, 2024-02-11:
@danger355 Yes, I believe that is true. The AP settings are if you have no other network, Ethernet or WiFi, available. If you have an Ethernet connection it will try to use that if you do NOT have networkless set to true. If you have it set to true, it looks to try to set up the hostapd and dnsmasq files to become the AP, with the static_ip file to disable the use of DHCP to pull an IP address from a regular network. 
What are you trying to do? What network resources do you have? Obviously a PoE switch, and the cameras are Ethernet on another network? Where are you trying to connect in the Pi to talk to the cameras?

danger355, 2024-02-11:
> What are you trying to do?

I'm working on a security system for some relatives that has some raw land - meaning no available internet - and they can sometimes go months without visiting. I say 'raw' but the are slowly building it out, and have increasing amounts of tools/machinery there. They're looking for a way to be able to record anything that might happen with a way to look at said recordings if they notice anything different or missing.

On a high level: There's no internet and also no power. There is, however, a few solar panels and batteries I hooked up for them a while back. I'll be using a 12v Pi Hat to power the Pi. This and the cameras will be connected to the DC-Powered PoE Switch... hopefully this will be enough to power the number of cameras they want...

> What network resources do you have? Obviously a PoE switch, and the cameras are Ethernet on another network? 

'network' as in internet? or as in 'devices' in general? The Pi, the PoE switch, and PoE cameras are all I plan to use; nothing else.

> Where are you trying to connect in the Pi to talk to the cameras?

The Pi and all the cameras will be hard wired to the PoE switch.

...

Next (not that you asked)... I'm going to try and use a USB GPS dongle to supply the time. Fingers crossed...

starbasessd, 2024-02-11:
@danger355 If it were me, I'd run a Pi3 or Pi4 with RaspberryPiOS Buster or Bullseye, run a DHCP server service like ISC DHCP. GPSD and Chrony for a time server, and have the expand-ability to add a wifi to cell modem adapter and other possible features. The Pi4 has a PoE hat, but you could get something like this https://www.amazon.com/UCTRONICS-Raspberry-Ethernet-Chromecast-Compliant/dp/B0876N3PMZ for the Pi3 or Pi4 to get PoE. I've used similar before.

danger355, 2024-02-11:
@starbasessd So there's no way to have it not look for internet while still maintaining basic network functionality?

> The Pi4 has a PoE hat

I have two Pi4's and also a PoE hat, but I can't get the 4 running for some reason, and opted to keep the Pi (whichever one I end up using) powered directly from the batteries instead of PoE just to make sure I don't approach the power budget of the switch.

Using PiOS, would MotionEye just be an installed program? Will I be able to access the dashboard via some network connection like MEyeOS?

starbasessd, 2024-02-11:
>So there's no way to have it not look for internet while still maintaining basic network functionality? 

It is not a scenario anticipated in motionEyeOS. 
Since it is web based interface, it assumes you either have a network you can connect to, or not. 
If you  don't have a network it can support just a set up as an access point. 
It never looks for the internet, unless you were going to use GDrive or DropBox, but both of those features are broken due to the providers breaking things. 
It IS a web based system to talk to  IP Cameras and administer its functions. 
To that end, it supports getting its network settings from a network DHCP server, or using certain files to set static info, or set itself up as an access point /dhcp server. 
You can just use the static_ip.conf file and not use the other AP files, but you need to also set the cameras to static IP addresses however that is accomplished. 
It is also theoretically possible to modify the dmsmasq.conf file to use the eth0 interface instead of wlan0 or ap0 interfaces, thus acting as the dhcp server for the ethernet side.
Unfortunately, being a manual setup on a very stripped OS, it is not as flexible as RPiOS Lite...

>Using PiOS, would MotionEye just be an installed program? Will I be able to access the dashboard via some network connection like MEyeOS?

Yes, with the exception of the port used ( http://ip_address:8765 vs just http://ip_address ) and a few minor tweaks, the setups give the same webgui. You can get the install instructions [here](https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye).

danger355, 2024-02-11:
@starbasessd I've been trying to use multiple versions of PiOS Lite ever since you mentioned it, and it keeps failing around installing python/motion

I really appreciate your help thus far, but my problem is I don't have a ton of experience with .conf files, linux in general, or cli for that matter; most of the time I can google my way through an issue, but not with this - not yet anyway, but I'm determined.

I think the best action here is to continue trying to get MEyeOS running. You mentioned it's possible to use static_ip.conf or dnsmasq.conf to achieve what I'm looking for. I've played with them (mostly dnsmasq) a little with no results. For example, I changed `interface=wlan0` hoping it'd dole out IP Addresses to anything plugged into the switch, but that seemed to kill the wap.

I realize this is a big ask, but could you help me out with the specifics of this? A small btw: I posted about this project on reddit and others are looking for a similar solution, so I'm sure it will go far beyond helping me. I'm also willing to go through the tests and report back as long as necessary.

Some clarification: this camera system will remain unattatched to any internet connection, and will be accessed locally when my relatives visit their land only if needed, aka: they will connect to it and review footage if they notice something wrong/broken/stolen/etc, and save footage if needed.

danger355, 2024-02-16:
@starbasessd Success!

So your last post might have been the kicker. The official MotionEye install instructions page links [here](https://github.com/motioneye-project/motioneye/wiki/Installation) right up top, and mentions Python3, but without your last post, I'd have not known that Python2.7 was required, and that the OS had to be Buster or Bullseye. It does look like MotionEye is developing a version that uses Python 3, but that's for another day.

I also happened by [this](https://forums.raspberrypi.com/viewtopic.php?t=193770&sid=1466d7bcdff1d6e6455d9f4740a22e3e&start=25#p1215605) post that enabled me to do exactly what I needed: access the Pi (and MotionEye) over the Pi's AP while also handing out local IP Addresses to all IP Cams.

I've since gotten the GPS working also thanks to your recommendation about Chrony and [this](https://austinsnerdythings.com/2021/09/29/millisecond-accurate-chrony-ntp-with-a-usb-gps-for-12-usd/) how-to, and also an SSD plugged into a USB port.

It sometimes takes a bit for the GPS to find satellites, and the SSD isn't hot pluggable and I can't replace it with another one without editing the UUID at the moment... but it does work, it's reliable, and will reboot when power is available by itself. I'm currently making an image of the SD Card, just to be safe; I nuked my original working version by removing it from a Pi3 and placing it in a Pi4 to try and have the USB3 ports... that was a bad idea. Speaking of, I couldn't get all this working on a Pi4. Will probably try this again along with the Python 3 version whenever it gets released.

About your final piece of advice about how fast the storage will fill up, I agree. I'm going to try and play with settings to have a small video size that's still usable and utilize masking to the max, or better yet just use images. I will replace the test drive I have now with a larger one, probably a 4TB.

Again, thank you!

Fenestrane21, 2024-07-17:
> @danger355 : As to motionEye on RPiOS, it HAS to be Buster or Bullseye (Legacy) it will NOT work with the latest Bookworm. Period. MotionEye REQUIRES Python2.7 and will NOT work with Python3. You have to install as root ('su -' or 'sudo su') I think you need to consider finding what we in the Ham world call an 'Elmer', a local guru to help you. I and many others have documented many how-tos, and there are a lot of documented solutions for lots of issues, for both [motionEye](https://github.com/motioneye-project/motioneye/wiki) and [motionEyeOS](https://github.com/motioneye-project/motioneyeos/wiki) here. Some of the standard motionEyeOS solutions will not apply to you as your use-case was not considered. The AP solution was designed for things like the PiZero (no ethernet) or Pi3 & 4 with WiFi but no laptop or PC connection. My recommendation of motionEye on RPiOS or similar lets you have the flexibility of using both ethernet & wifi. You really need to also consider how often the family will be checking the site, as video recording, especially at higher resolutions, frame rates will eat HD space rapidly. If you allow the storage to fill, it will cause the system to crash on either setup... I kill a 1TB drive in 3-5 days with 6 cameras around a house. I have custom scripts to clean it up. If I wasn't able to access at least twice a week, and didn't have the scripts, and needed to keep them between visits, I'd be in trouble... If I were a dev, I would reconsider some of these features for the next version, but I am not seeing any work progress by them, and have been winding down my support.

Hi starbasessd, I hope you can help me out, been trying this for a while now. I want to make a dashcam with motioneyeos and a raspberry pi zero 2 w, have hooked up a rev 1.3 cam with it. Got everything up and running after a few months, but with a wifi connection of my home router, so my problem is that there is not any wifi connection in the car, so how can i create it its own wifi acess point? I just want to have motion detection and videos being saved to the sd card i have attached. I dont think the above files are compatible with a pi zero 2 w. Seen you help a lot of others, so decided to approach you

Thanks a lot!

Fenestrane21, 2024-07-17:
@starbasessd 

Hi starbasessd, I hope you can help me out, been trying this for a while now. I want to make a dashcam with motioneyeos and a raspberry pi zero 2 w, have hooked up a rev 1.3 cam with it. Got everything up and running after a few months, but with a wifi connection of my home router, so my problem is that there is not any wifi connection in the car, so how can i create it its own wifi acess point? I just want to have motion detection and videos being saved to the sd card i have attached. I dont think the above files are compatible with a pi zero 2 w. Seen you help a lot of others, so decided to approach you

Thanks a lot!

Demetriao, 2024-07-25:
> > Yea that is true, but jawsper edited the version to work on the pi zero 2 w.
> 
> Where do I find that version?

https://github.com/jawsper/motioneyeos/releases