motioneye-project/motioneyeos #1419: thank you!!!

mega-hz, 2018-03-03:
Hi Ccrisan,

i used a very old version of motioneye running on a pi2 ,
after a period of about 3 years the hdd died, on it the whole motioneye installation.
So i found your motioneyos and set it up with another hdd.
I AM AMAZING to see, how easy things go today!
I just want to say THANK YOU for your great work on this!!!

I just have one question:
Is it possible to write the motioneyeos-image directly to the hdd and set the boot-dev to sda1?
So the whole system runs completly of the hdd instead of sd-card?

If so, how?

regards,
Wolfram.

ccrisan, 2018-03-04:
You can't have the OS boot from an external HDD. You shouldn't however care about the SD card as it's mostly read-only. If you're worried about log files being written to the SD card, you can move them to RAM (see [here](https://github.com/ccrisan/thingos/wiki/Logs) more about logs in motionEyeOS).

closed 2018-03-04 by ccrisan

mega-hz, 2018-03-04:
with /dev/sda1 instead of mmcblkp1 in the config.txt (on sd card) the old installation of motioneye run only on hdd! 

why shouldn't run motioneyeos as same?
please explain...

ccrisan, 2018-03-04:
@mega-hz because motionEyeOS a bit different than a general purpose OS. Many parts of the OS are particularized and rely on specific paths and devices. Of course you can tweak it to run from a HDD, but your tweaks would be overwritten by a firmware update.

mega-hz, 2018-03-04:
ok thank you, hope the sd-card will live long.

jasaw, 2018-03-05:
@mega-hz Flash chips typically can survive 10k and above erase cycles, and SD cards have wear-leveling algorithm to distribute the flash wear across the entire card. On top of that, if you move the log files to RAM, there's no write activity to your SD card, it will survive longer than your traditional spinning hard disks. There's no need to worry too much about the SD card dying.