motioneye-project/motioneyeos #1420: Static_ip

berniemci, 2018-03-04:
Hi, I created a static_ip.conf file as per the instructions and the file has 3 lines:
static_ip="192.168.1.105/24"
static_gw="192.168.1.1"
static_dns="8.8.8.8"

When I have tried using this today with a Raspberry Pi 1 and 3 using the respective new releases from 20180224. 
I get the following error:
" ip: an inet address is expected rather than "192.168.1.1
failed
I have checked the IPv4 gateway and it is definitely 192.168.1.1. I'm not sure what to do next?

berniemci, 2018-03-04:
Just to confirm I have loaded the static_ip.conf prior to initial boot.

ressit, 2018-03-04:
Try to connect first with default DHCP, if supported by your router, and change the setting
using the web interface "Network -> Manual  (static IP)".

berniemci, 2018-03-05:
I can connect to the default DHCP and can then change to a manual (static IP), that isn't a problem. I just cannot get the static_ip.conf file to do this for me by following the instructions.

ccrisan, 2018-03-05:
Please attach the static_ip.conf file and your boot.log file.

berniemci, 2018-03-05:
Dear Calin, it works now. Just tested it. By reading issue #876 I learned about saving the notepad ++ file as a unix lf file. And it worked beautifully. I previously had it on a Windows setting (purely from being a noob.)

The system is great and well done on building such a great system.

closed 2018-03-05 by berniemci