motioneye-project/motioneyeos #1421: Always read wpa_supplicant.conf from boot partition on bootup

jasaw, 2018-03-04:
To improve user experience, it would be better to read `wpa_supplicant.conf` from boot partition whenever the file is updated. This prevents the need to reflash the SD card if wpa_supplicant.conf was missing or wrong at **first** boot.

As @ccrisan suggested: a complex but viable solution would be to keep a hash of the boot variant and therefore be able to tell when it has changed. If the boot variant has changed, we use it to overwrite the data variant. And this mechanism could/should be applied for all config files that support "boot provisioning".

A few examples to describe how this would work:
1. Initial boot - copy wifi config from boot and save hash.
2. Wifi config changed via web interface - hash stays the same, do not copy from boot.
3. Updated wifi config in boot partition - hash is different, copy wifi config from boot and save hash.

Original discussions here: https://github.com/ccrisan/motioneyeos/issues/876

ccrisan, 2018-04-17:
I must admit it's definitely a simpler solution than the one originally proposed by me :) And does the same job.

jasaw, 2018-06-04:
@ccrisan There's one issue with deleting the `/boot/wpa_supplicant.conf` file after it's copied to `/data/etc`. The boot partition is normally mounted read-only with the option of mounting with write permission depending on os_debug flag. I can't think of a cleaner way of doing this, except to remount boot with rw just before we delete the file.

This is what my S35wifi looks like:
```shell
if [ -f $boot_conf ]; then
	cp -f $boot_conf $conf
	grep -E "/boot .*ro[\s,]" /proc/mounts
	RO=$?
	test $RO == 0 && mount -o remount,rw /boot
	rm -f $boot_conf
	test $RO == 0 && mount -o remount,ro /boot
fi
```

Another issue is when users upgrade firmware, their `/data/etc/wpa_supplicant.conf` will be overwritten by `/boot/wpa_supplicant.conf` if the file exists in `/boot`, which means they may end up with a misconfigured wifi (if boot version differs from actual). One workaround is to remove `/boot/wpa_supplicant.conf` in S14postupgrade, which again involves remounting boot with rw.

Thoughts?

ccrisan, 2018-06-20:
@jasaw I agree that if we want to implement it this way, we should temporarily remount the partition read-write. Can we remount it read-only back again without a reboot though?

alphanumeric007, 2018-08-07:
Why does it endlessly reboot when no network is found, in the first place? That is so frustrating and I just don't see why its like that? I had a monitor keyboard and mouse connected on first boot and all I got was endless reboots? Why can't you just go in after it boots up and setup your WIFI? IMHO having to edit and add a wpa_supplicant file is not the answer. Its way to easy to have an editing error, especially if your doing it on a Windows PC.
At the very least have it prompt for your wifi credentials on boot up. I ended up plugging in a USB hub with an ethernet connection so I could get in and setup a wifi connection via the motioneye setup menu. That's is not ideal. IMHO the user experience needs to be improved.

jasaw, 2018-08-08:
@alphanumeric007 I believe rebooting when no network connection behaviour is to workaround bad wifi drivers/hardware that only recover after a reboot. You can disable this from the web UI.

Why do we need write the wpa_supplicant.conf file?
That's because the target device may be headless, and it's unreasonable to expect all users to plug in a monitor and keyboard to a headless device. Also, if the user is provisioning multiple cameras, it's easier to drop the same wpa_supplicant file into multiple devices, rather than typing it in from the console (which is prone to typos too).

I agree with you that the user experience needs to be improved, but I do not agree with your approach of prompting user to enter wifi credentials at boot time. Like I mentioned before, I believe setting up a wifi access point when wifi is not configured gives a better user experience.

alphanumeric007, 2018-08-08:
@jasaw  Ok, thanks for the response. 
You can't disable rebooting if you can't get in though? That's impossible with a stock image on first bootup if you don't plug in an ethernet cable. Which is no easy feat on a Pi Zero.
I get why it does it, its a web camera that's likely headless. But actually setting it up headless is IMHO a PITA. The first time boot up shouldn't allow endless reboots, prompt for something, give us a pause to do something? Especially if you do happen to have a keyboard mouse etc plugged in. 
 How do you setup a wifi access point if the wifi is not configured? That's not auto is it? I'm going to go back and reread the whole thread, once I get a cup of java in me. But right now that statement just confuses me? Too early in the morning for this stuff lol. It's not even 6AM yet here. :)