motioneye-project/motioneyeos #1426: IR Light Ring Not Working

jbirmingham704, 2018-03-09:
I installed MostionEyeOS on a Raspberry Pi 3.  Everything is running is working fine.  I bought and installed an IR light ring, I programmed an action button using the Simple GPIO Example on the Action Button page of the wiki but I included it below for convenience.  However when I click the Light On button, nothing happens.  The pin out is:

5v > pin2
Gnd > pin4
GPIO > pin12 (GPIO 18)

light_on_2:

#!/bin/bash

GPIO=18
test -e /sys/class/gpio/gpio$GPIO || echo $GPIO > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio$GPIO/direction
echo 1 > /sys/class/gpio/gpio$GPIO/value

I'm just looking for any suggestions on what I may have done wrong or other ways to test the IR light.

Thanks in advance.

gitgoldberg, 2018-03-09:
Which IR light ring product are you using ?
Have you tried powering it from a 5v source just to check that it works independently of the Raspberry Pi ?
What current is it expected to draw (does it exceed the specified current draw for a GPIO pin ?)
It's not by any chance an I2C device, such as "Bright Pi' ?

I've successfully used the light_on method you describe above, but with the GPIO pin driving a relay module, and the LED module then powered from a separate supply independent of the Raspberry Pi's 5v supply.

jbirmingham704, 2018-03-11:
It's a Lisiparoi IR light ring. I haven't tried to power it independently, i'm not sure how to do that.

gitgoldberg, 2018-03-12:
In the Lisiparoi instructions at lisiparoi.com - under the heading 'Wiring up to the Raspberry Pi' is described a simple test for the Lisiparoi, using the Pi's +5v and GND, and the 3v3 pin to switch the Lisiparoi on.  The Lisiparoi draws around 600mA so you will need to power the Pi from at least a 2 amp, preferably 2.5amp 5v power supply.

When the Raspberry Pi is switched on, the Lisiparoi should light up with this configuration, although you might need a smartphone camera to detect the IR light as described under 'Powering it all up' further down the list of instructions.

If this simple test doesn't work, there is probably something wrong with the Lisiparoi, or it's not wired-up correctly.

Once you've established that the Lisiparoi is working, you then need to move its GPIO connection to your chosen GPIO pin on the Pi, to test it under program control.

MrGJones, 2019-03-11:
Hi, I'm having a similar problem with this light ring.. the lisiparoi.com site appears down so I am unable to locate information on how to set this up... is there anything i need to configure in motioneyeos to enable it? i have 5v, Ground and the GPIO 10 cables connected but im having no joy?

jbirmingham704, 2019-03-11:
Turns out the light ring was defective, I exchanged it for a new one and it worked as described.

closed 2019-03-11 by jbirmingham704

Brewj, 2020-04-27:
Hi there, I've stumbled across the attached trying to get my LED ring to work, sadly it appears to be defective.
[Lisiparoi LED Ring.pdf](https://github.com/ccrisan/motioneyeos/files/4541718/Lisiparoi.LED.Ring.pdf)