motioneye-project/motioneyeos #1435: MotionEye with StretchDesktop

gizzburn, 2018-03-14:
Hi new to motioneyeos, I have installed it on a pi 3 and its working great, what I cant seem to find out is how to install motioneye with stretch desktop already installed, I want to install a small 3.5 led screen on my pi3 so I can see what the camera is seeing in motioneye.

I cant seem to find out if this is possible with stretch already installed how to add motion to the system, is this possible and if so any links to a how to, as I am new to all this but have tried a search and reading up but drawn a blank.

Mick

Moonbase59, 2018-04-07:
Firstly, let me say that I have no idea how to do that. ;-)

But if you’re open for other ideas, here is what could be done and what I do in my home:

1. I seem to remember that you could use a small HDMI screen (camera monitor) and simply hook it up to your Pi. Might bust the budget, though.

2. My solution: I currently use only one camera, so I opted for not trying to change a whole OS (which MotionEyeOS is) but install it on a Pi Zero W. Cost me around 20-something Euros, including the official case, SD card, the Pi Zero W and a mains adapter (not the camera which I had). I then let it stream on port 8081 and thus can watch the camera stream on every PC in my home, and even on my phone (which has a display) and my TV. Maybe this would be an option for you? (Be warned: You wouldn’t be able to get a high-fps high-res stream using only a Pi Zero, but 1 fps @ 1280x720 was enough for my purposes. And you could go for "Fast Network Camera" mode of course.)

jasaw, 2018-04-09:
@gizzburn If you want to install motion and motioneye on Raspbian, follow the instructions here: https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

What you want to do should be possible.
1. Assuming you're going to use an LCD similar to the [waveshare](https://www.waveshare.com/3.5inch-RPi-LCD-B.htm). You follow the [waveshare installation guide](https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(B)) to get your LCD working. You should see your desktop on the LCD when it's working.
2. Follow the [motioneye installation guide](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) to get motion and motioneye on Raspbian.
3. Auto-start Chromium web browser in kiosk mode at boot up:
     `sudo nano ~/.config/autostart/autoChromium.desktop`
     Add the below contents to autoChromium.desktop:
     ```
     [Desktop Entry]
     Type=Application
     Exec=/usr/bin/chromium-browser --noerrdialogs --disable-session-crashed-bubble --disable-infobars --kiosk http://localhost --incognito
     Hidden=false
     X-GNOME-Autostart-enabled=true
     Name[en_US]=AutoChromium
     Name=AutoChromium
     Comment=Start Chromium when GNOME starts
     ```
     Change the `http://localhost` to point to your camera stream. You may need to create a html file and embed the camera stream inside, I don't know. I'm not a web developer.
4. When you're happy with how the camera stream looks on your 3.5" LCD, you can disable screen blanking and cursor to make it look good.
   * Hide taskbar when idle. From the desktop, right click on the taskbar to configure it.
   * Disable screensaver and disable screen blanking (power saving). Edit __/etc/xdg/lxsession/LXDE/autostart__ and __/etc/xdg/lxsession/LXDE-pi/autostart__.
     ```
     sudo vi /etc/xdg/lxsession/LXDE/autostart
     sudo vi /etc/xdg/lxsession/LXDE-pi/autostart
     ```
   * Add a # to the beginning of the line to commend it out.
     ```#@xscreensaver -no-splash```
   * Next add these lines underneath the screensaver line to disable screen blanking.
     ```
     @xset s off
     @xset -dpms
     @xset s noblank
     ```
     __xset s off__ disables the screen saver, __xset -dpms__ disables the DPMS (Display Power Management Signaling) and __xset s noblank__ tells to X server to not blank the video device.

   * Disable cursor and screen blanking. Edit __/etc/lightdm/lightdm.conf__.
     Add this to your file: ```xserver-command=X -nocursor -s 0 -dpms```

5. Other things to consider:
   * You may want to make the root filesystem read-only to reduce the probability of filesystem corruption in the event of sudden power loss. Look into `overlayfs`, which will work with the desktop.