motioneye-project/motioneyeos #1439: Fast network camera breaks add a camera

Systemgeek, 2018-03-16:
I have a raspberry pi zero w running:
> motionEye Version | 0.38.1
> Motion Version | 4.1.1
> OS Version | motionEyeOS 20180224

I have a central camera server running on an Tinkerboard with:
> motionEye Version | 0.38.1
> Motion Version | 3.2.12+git20140228
> OS Version | Ubuntu 16.04

It took me a long time to get the remote camera added to the central camera server.  Finally I got it to work after disabling Fast Network Camera.  Is this a bug or should it really be disabled if adding a camera to a central camera server?  I ask because all that I read sounded like you should leave it on.

Thanks,
Louis

ccrisan, 2018-03-16:
It's written in the [Fast Network Camera](https://github.com/ccrisan/motioneyeos/wiki/Fast-Network-Camera) wiki page that this mode only works with the CSI camera. You can't add anything else and the reason is that the entire backend (which is normally *motion*) is replaced by *streamEye* which only works as described.

closed 2018-03-16 by ccrisan

Systemgeek, 2018-03-18:
Maybe I am not explaining myself right.  The camera itself is one that I purchased as a CSI camera with a Raspberry PI 2 a while back.  Its now plugged in to a Raspberry PI Zero W via the CSI cable.  The Tinkerboard does not have any cameras directly connected to it nor will it ever have a direct cameral plugged in to it.

So according to that wiki page the Raspberry pi zero w should be using Fast Network Camera since it will be using the GPU on the RPI Zero W.  Or am I not understanding how this works.

Louis

ccrisan, 2018-03-18:
@Systemgeek you appear to have understood it correctly. What is unclear to me is what does the Tinkerboard have to do with Fast Network Camera?

As far as I understood, your Tinkerboard runs Ubuntu + motionEye, where your Fast Network Camera was added as a Network Camera. Am I correct? If yes, what seems to be the problem?

Systemgeek, 2018-03-19:
So as I understand it when I try adding a remote camera to the tinkerboard the remote camera (running on RPI zero W) should be configured for Fast Network Camera.  However, when I tried that MotionEye on the tinkerboard was unable to find the camera on the RPI.  By accident I had clicked off the Fast Network Camera and suddenly the tinkerboard was able to see it and add the camera.

Since then I have tried changing the Fast Network Camera setting on.  When I do that the tinker board losses connection to the RPI.  I turn it off and it works again.

But this is counter to what I understood the configuration should be.

Louis

ccrisan, 2018-03-19:
> So as I understand it when I try adding a remote camera to the tinkerboard the remote camera (running on RPI zero W) should be configured for Fast Network Camera. 

Nope. If you want to add your Zero camera to your Tinkerboard as a *Remote motionEye Camera*, then you must *not* enable Fast Network Camera on your Zero and leave it as default.

If on the other hand you want to add your Zero camera to your Tinkerboard as a *Network Camera*, then you can add it in both modes, each one with its advantages and disadvantages.

See [Usage Scenarios](https://github.com/ccrisan/motioneyeos/wiki/Usage-Scenarios) for more details. The term "hub" is used for the former method, while "central server" refers to the latter one.