motioneye-project/motioneyeos #1440: Possible to have option for dyndns? type no-ip.org?

tigernero79, 2018-03-17:
it would be handy to have integrated option to have the dyndns on the motioneyeos system. imagine a Wi-Fi connection with a cell phone that acts as an access point whose IP would change with each connection. the dyndns would proceed to recall ip just remotely by name and not ip. in order to always have a mobile cam. It can be done?

Chiny91, 2018-03-17:
It has been on the "to do" list for some while:
https://github.com/ccrisan/motioneyeos/wiki/Roadmap

It is reasonably simple (perhaps not as simple as having it built-in to motionEyeOS) to set it up externally.

psts, 2018-03-18:
For noip.com, add this to your crontab: 
0 * * * * curl -u ‘your@email.com’:yourpassword "http://dynupdate.no-ip.com/nic/update?hostname=yourhostname.ddns.net"

tigernero79, 2018-03-18:
Perfect I'll try as soon as I get rpi3 b + and more zero wh. I have no ip in fact

Moonbase59, 2018-04-01:
I let my router do it for me (an AVM Fritz!Box), for both IPv4 and IPv6. Using [nsupdate.info](https://nsupdate.info/).

closed 2019-08-29 by tigernero79

cuervopunk, 2020-08-30:
> For noip.com, add this to your crontab:
> 0 * * * * curl -u ‘[your@email.com](mailto:your@email.com)’:yourpassword "http://dynupdate.no-ip.com/nic/update?hostname=yourhostname.ddns.net"

worked for me, thanks!