motioneye-project/motioneyeos #1441: Sequential File Name?

rdb9879, 2018-03-17:
Hello. I am using motioneyeos as a dash cam project in my car. Since my camera/pi will be mobile, I decided it was best to have the rasp pi zero w generate its own wifi hotspot by providing hostapd.conf. This means that there is no true time source, and every time I restart my car, the time is set to default Jan 1, 1970. This makes the video files appear on my sd card in a scrambled order, since they all more or less have the same file name/create date.

Is there any way to make the video file names just have a simple sequential index/number? Like when it decides to record a new video, it will read what the highest number that exists in the destination folder, increment it by one, and use that number as the new filename?

I am open to other solutions as well. I'd just like a good way to determine which video is the most recent.

rdb9879, 2018-03-18:
Nevermind on this. I figured out how to do what I need using "Run Command After ..." and scripts. Thanks anyhow.

closed 2018-03-18 by rdb9879

joshferns, 2020-04-11:
> Nevermind on this. I figured out how to do what I need using "Run Command After ..." and scripts. Thanks anyhow.

Hi @rdb9879 I'm having a similar issue at the moment, could you please let me know how you ended up solving this?

Thanks.

lucianaohira, 2020-12-01:
Hi all!!! We are also having the same issue... Hey @rdb9879, help us, pleeease! We have no idea about how to run scripts... :(

lucianaohira, 2020-12-01:
Hey, hi all! Hey @joshferns!
In the field "Image File Name" (or Movie File Name), we just wrote %v for saving the still images at a number sequence like 01.jpg, 02.jpg and so on.

joshferns, 2020-12-02:
Thanks a lot @lucianaohira will give this a shot.