motioneye-project/motioneyeos #1443: Cant connect after 5 minutes of running

LimeSpysk, 2018-03-18:
So if i start the pi it works fine, i can connect and it works but if i leave it running for a few minutes i loose conection and i cant connect to it anymore

amolpravar, 2018-03-24:
Yes I can confirm this, I updated the Motion Eye OS live when the Pi was switched on with Motion Eye OS running( No NOOBS) and the Pi get's even disconnected sometimes.

Octa604, 2018-04-01:
I've been having this issue ever since updating to the newest release. It seems that as soon as I'm in the UI it crashes. Going to downgrade back to 20171008 and will report back.

Moonbase59, 2018-04-01:
Hm. I’m using motionEyeOS 20180224 on a Pi Zero W with 2 fps @ 1280x720 (Raspi Cam, official case, taped into a sunlit window). It runs a 1 fps 720p public MJPG stream, plus takes a still every minute for time lapse video. Typical load average is 0.18/0.32/0.30 up to 0.53/0.47/0.39 and it now runs for about 9 days, according to uptime.

Octa604, 2018-04-01:
> I've been having this issue ever since updating to the newest release. It seems that as soon as I'm in the UI it crashes. Going to downgrade back to 20171008 and will report back. 

Downgraded my cameras to 20171008, Had the exact same issue pretty quickly.  I then gave each camera it's own HTTP port and it seems to be stable now. Could this possibly be caused by too much traffic through the port they are on?