motioneye-project/motioneyeos #1446: Gmail with "less secure apps"

freddy64, 2018-03-20:
Hello,
Can I now disable the " Less Secure Apps” in Gmail and still get notifications?
I needed to permit logins from "Less Secure Apps" to get notification from motionEyeOs but google tells me this is a security problem. So I'm wondering how much of an issue this is.
I'm using motionEyeOS 20180224 with Raspberry Pi Zero W as security cameras.
Thanks

mgcdrd, 2018-03-23:
The technical answer is it is a security risk and caution should be taken when disabling this setting.  If you have an Android phone and have Gmail enabled on it, you get notified when something new logs into your account.
I have use my Gmail account with other "less secure apps" and don't have issues.  My guess this setting has a lot to do with the secure connections and Google's certificate transparency movement.

freddy64, 2022-04-07:
In my security settings on my Google account, the less secure apps end on the 30th of may 2022. This means I will no longer get email notifications. I am currently using motionEyeOS 20200606. Will there be a new version OS to enable email notification?

starbasessd, 2022-04-07:
Unknown at this time whether GMail will still allow the current ability to send emails through their servers, which is NOT an  'insecure app'. The primary affected systems will be things like GDrive and other functionality, and I personally have not had any response back as to whether the AppToken method used for the current GDrive will be considered 'insecure' or not.

RDPythonSoft, 2022-05-05:
Google will even left out the setting possibility in the "Google account settings".
"Allow lower security apps: ON
As of May 30, 2022, this setting will no longer be available".
Is there an urgent solution for Motioneye needed??

freddy64, 2022-05-06:
As for now, the 6th of may 2022, I'm still getting email notifications with picture attachments on my mobile phone and pictures/movies are still being uploaded to my Google Drive.

RDPythonSoft, 2022-06-08:
The switch-off by Google is a fact.
When activating a test e-mail in Motioneye notification the Google error code 5.7.8 (username and password not accepted) is given.
"Allow lower security apps" setting is not available anymore in the Gmail account-settings.
Is there a solution to still use Gmail?

starbasessd, 2022-06-08:
It was ignored. Creating the pass key fails in the email notifications, too.

freddy64, 2022-06-08:
So it finally happened, no more email notifications from Motioneye, since the "Allow lower security apps" setting is not available anymore in the Gmail account-settings. I hope there is a solution.

freddy64, 2022-06-13:
I tried switching to Outlook.com myself, but I didn't work for me. First, I received emails with picture attachments in the 'sent items' as opposed to my 'inbox' so I wouldn't get notifications on my mobile and then Outlook.com blocked my account considering there were too many emails or spam. Not sure what i did wrong.