motioneye-project/motioneyeos #1451: How many Cameras

gizzburn, 2018-03-24:
I am running Motioneyeos 4.1.1 OS 20180224, its on a Pi3, I have 3 cameras setup and running, it all seems a bit hit and miss when they capture any motion. One of my cameras will not capture any motion at all.

I have two ip cameras, one is the ieGeek Outdoor IP Camera,Full HD 720P IP66 that works ok the other is a ieGeek 1080P , this one will not work on any settings for motion capture.
My other camera is the Kuman Camera Module for Raspberry Pi 3 Model B B+ A+ 2 1 5MP 1080p OV5647 Sensor HD Video Webcam Night Vision Camera SC15 , which works very well with motioneyeos.

Am I asking to much  from a PI3 with this setup trying to capture from all sources.? If so would an ODROID-XU4Q or similar be better, or get more Pi3's

MJ.

twiggyr, 2018-03-29:
I tried a XU4 vs Pi3 and found the XU4 doesnt do hardware x264 encoding of the video's which leads to the saved files being jerky and missing frames .   A Haswell or later NUC will probably give better performance but at increased cost.

gizzburn, 2018-03-29:
Thanks twiggyr, I now am getting somewhere all camera are now reporting any motion and the main camera , camera2 that would not work is now responding as well. It seems like its a trail and error setting to find what works and what does not..
Anyway its working quite good now with my three cameras, just trying to read up now on how you can view your feeds by remote when away from home.

MJ.

Chiny91, 2018-03-29:
@gizzburn _trying to read up now on how you can view your feeds by remote when away from home_

I suspect many brave people simply knock a hole in their defences and forward the relevant port from their router to their R Pi.   The less brave might use OpenVPN on their router and thence to their R Pi.   Villains can then be further deterred by firewalling your R Pi from the net altogether (although you need an ntp server inhouse).   Unless you have a static IP address, sorting dynamic DNS is required (on motionEyeOS wish list).

jasaw, 2018-04-18:
@gizzburn Are you happy to close this issue?

gizzburn, 2018-04-18:
Yes all sorted now thanks

closed 2018-04-19 by jasaw