motioneye-project/motioneyeos #1452: RPi Zero W - Conifuring in AP mode

intel2004, 2018-03-25:
i think i found a bug in the configuration...
i need to create a cam with raspi zero w, if i connect directly with wifi it's ok.
i found in wiki to configure with hostapd to create an access point, i need that function because the cam is far away from internet!
but if i configure a clean image with only hostapd.conf,dnsmasq.conf but it always reboot because it can't find wired network...OF COURCE IT CAN'T!
i try to put os.conf in boot partition with os_networkless="false" but it can't stop.
any idea????

intel2004, 2018-03-25:
if i configure first wpa_supplicant before and next hostapd it always connect to configuration of wpa_supplicant and not create the AP if i already remove the file from boot directory...

intel2004, 2018-03-25:
if i flash a new image, put the hostapd and dnsresolve configuration file and bootup...
searching for wired lan, then rebooting in loop.
on raccomandation for raspberrypi zero says to use image for first raspberry, i don't know if that is useful.
i already configure that now, but in strange way.
on SSH access if i try to edit or remove some file they say that i don't have permission even in root mode.
i have to mount ext2 partition on another system to do that?

ccrisan, 2018-03-25:
Set `os_networkless="false"` in your `/data/etc/os.conf`. You'll need a Linux PC for that. Otherwise, you could provide an `os.conf` file on your boot partition, but you'll need to provide the entire file. You can use [this](https://github.com/ccrisan/motioneyeos/blob/master/board/common/overlay/etc/os.conf) as an example.

ccrisan, 2018-05-20:
@LordFinland please attach your `boot.log` and `messages.log` files.

LordFinland, 2018-05-21:
I will asap (At school currently). I got around the issue by creating the WiFi AP in Raspbian lite according to the official documentation. I then installed Motioneye as a package separately. I will retry my earlier method and upload the log files this evening.

LordFinland, 2018-05-21:
So, having followed instructions found [here](https://github.com/ccrisan/motioneyeos/wiki/WiFi-Access-Point/177f7d30dc3e564baa914109f84f5653211aa45e) Since there is no /data/etc/os.conf file, or directory, I changed the file  /etc/os.conf and added [os_networkless="true"] to the end. I also added the two configs as per your instructions all before first-boot.

Fortunately however, I didn't encounter the same problem as yesterday. The system boots up just fine, but hostapd encountered errors with each entry in its configuration file. Maybe the indentation? So adding os_networkless to the os.conf file works just fine.
[boot.log](https://github.com/ccrisan/motioneyeos/files/2023688/boot.log)
[hostapd.log](https://github.com/ccrisan/motioneyeos/files/2023689/hostapd.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2023690/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/2023691/messages.log)

LordFinland, 2018-05-21:
Indentation is not to blame it seems... I tried it (The hostapd configuration changed) again, this time with a completely different result.
[hostapd.log](https://github.com/ccrisan/motioneyeos/files/2023713/hostapd.log)

ccrisan, 2018-05-21:
@LordFinland if you could show us your `hostapd.conf` file, we could point out possible problems inside it :)

ccrisan, 2018-05-21:
My assumption is that there's a problem with newline characters (which could be Windows-style instead of Unix-style).

LordFinland, 2018-05-22:
Yes, probably. I created the .conf file on windows using notepad++.

` ssid=DRONEFEED
 wpa_passphrase=12345678
 wpa=2
 wpa_key_mgmt=WPA-PSK
 interface=wlan0
 channel=1
 driver=nl80211
 hw_mode=g
 ieee80211n=1
 ieee80211ac=1`

That's pasted from the conf file. GitHub doesn't support the file type it seems. Ill re-write the conf files and do the process solely on my parrot sec computer tonight. Maybe changing the format type in the editor to UNIX might have helped :-)

[image]

LordFinland, 2018-05-23:
I changed to UNIX encoding as from above screenshot. Everything works great now. AP runs fine and motioneyeos is natively installed. Yay. I would have never guessed what the issue was.... Sounds like my grade wont be a zero after all :-)

And as for @intel2004 you said you set "os_networkless" option in os.conf  to false? If you want the OS to be networkless, it should be true, not false right? At least I set it as true and it works. Secondly, with the latest motioneyeos build (Pi 0 W) you need to change the default os.conf file in the data partition under /etc/os.conf not /data/etc/os.conf (which doesn't exist). Lastly, changing hostapd configuration to use "n" network mode doesn't seem to work either, so stick with g.

ccrisan, 2018-05-28:
os_networkless should be set to true, as its name suggests. The file that should be edited is /boot/os.conf, before the first boot, or /data/etc/os.conf, if the system has been booted at least once.

LordFinland, 2018-05-29:
I looked for an os.conf file in /boot but it seemed to be non-existent. I could only find it in the data partition under etc. I tried creating a copy of the os.conf in the boot partition before first boot but it didn't help, the system seemed to apply the file from the data partition instead during first-boot.