motioneye-project/motioneyeos #1453: Broken config.txt after update

Instagit, 2018-03-25:
Hi,
after auto-updating to

````
motionEye Version | 0.37.1
Motion Version | 4.0.1+git2a140fa
OS Version | motionEyeOS 20171008
````

the config.txt on my Raspberry Zero was changed to this:

````
gpu_mem=128
gpu_mem_256=128
gpu_mem_512=128
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on
dtparam=audio=on
# Disable the ACT LED.
dtparam=act_led_trigger=none
dtparam=act_led_activelow=oninitramfs fwupdater.gz
arm_freq=700
gpu_freq=250
sdram_freq=400
over_voltage=0
disable_camera_led=1
````

Notice the misplaced *initramfs fwupdater.gz*. It wasn't possible to turn off the light until I fixed that.

Thanks for the work on motioneyeos. :)

ccrisan, 2018-03-26:
Thanks for reporting this. Is it reproducible? Can you briefly describe the steps to get to this state of `/boot/config.txt`?

Instagit, 2018-03-27:
All I did was check for updates through the web interface and install them. After that I noticed the light that I had switched off with

````
# Disable the ACT LED.
dtparam=act_led_trigger=none
dtparam=act_led_activelow=on
````

was always on. So I checked the config.txt and found the error.

Reproducing this will be difficult, because I don't even know which version I had previously installed or how exactly the config.txt looked. In any case, if you get more reports on device settings that don't work after an update, maybe this helps with identifying the cause.

Instagit, 2018-03-27:
https://github.com/ccrisan/motioneyeos/blob/e4b2abd652946cd7da674f710c104ec666770503/board/raspberrypi/overlay/usr/libexec/fw-prepare-boot#L3

I think the problem is there. When the config.txt doesn't have an empty line at the end, ````initramfs fwupdater.gz```` will be appended to the last config line.

Instagit, 2018-03-27:
This should work:

#!/bin/bash
````
config="/boot/config.txt"

if [ -z "$(tail -c 1 $config)" ]; then
    printf "initramfs fwupdater.gz\n" >> $config
else
    printf "\ninitramfs fwupdater.gz\n" >> $config
fi
````

ccrisan, 2018-03-27:
Yeap, that's the problem. It's now fixed upstream: https://github.com/ccrisan/thingos/commit/26cddbe840d1bc15fb2f5a7a3ddd7036ab132521.

closed 2018-03-27 by ccrisan