motioneye-project/motioneyeos #1460: Configuring wireless network: no device (RPI3)

VFR4EVER, 2018-03-29:
Hi there!

I have been struggling with this error and can´t understand what is happening in my RPI3.
I have the right version of OS, wired connection work fine, but wireless module seems to be inactive. 
What am I doing wrong?

Thanks for any help!
Cheers

ccrisan, 2018-03-29:
1. What OS version (full URL to what you have downloaded).
2. Are you 100% sure it's a Raspberry Pi 3 model?
3. Does your wireless card work with Raspbian?
4. I suppose the error you set as issue title is part of what you see in your `boot.log` file. Can we see the entire file?

VFR4EVER, 2018-03-31:
Here is the URL:
https://github.com/ccrisan/motioneyeos/releases/download/20180224/motioneyeos-raspberrypi3-20180224.img.gz

Positive! 100% sure

 I have installed raspbian and wifi is not working also... :(

I could see the error from the boot screen.

 So, something must be wrong with my RPi, right?

 Thanks

*De:* Calin Crisan <notifications@github.com>
*Enviada:* quinta-feira, 29 de março de 2018 20:52
*Para:* ccrisan/motioneyeos <motioneyeos@noreply.github.com>
*Cc:* VFR4EVER <delfimvfr@gmail.com>; Author <author@noreply.github.com>
*Assunto:* Re: [ccrisan/motioneyeos] Configuring wireless network: no
device (RPI3) (#1460)

   1. What OS version (full URL to what you have downloaded).
   2. Are you 100% sure it's a Raspberry Pi 3 model?
   3. Does your wireless card work with Raspbian?
   4. I suppose the error you set as issue title is part of what you see in
   your boot.log file. Can we see the entire file?

ccrisan, 2018-03-31:
There might be some hardware problems with your board. See if you can return it.

closed 2018-03-31 by ccrisan

VFR4EVER, 2018-04-02:
Thanks a lot. I ll try to figure out what is going on with my RPI..
Cheers