motioneye-project/motioneyeos #1461: date in the European format timestamp

tigernero79, 2018-03-29:
is it possible to choose the date in the European format under the item timestamp, then first the day then the month and finally the year? this can already be done for saved videos but not for live timestamp viewing on cam

ccrisan, 2018-03-30:
@tigernero79 where exactly would you like the time format to appear differently? Show us with a screenshot.

tigernero79, 2018-03-30:
simply instead of going out month and day, I would like to come out day and month, reverse year with days

[image]

now it goes 2018-03-31 I would like to go out 31-03-2018 in europe format with year at the end

ccrisan, 2018-03-30:
@tigernero79 in the Settings Panel, go to Text Overlay, choose Custom Text and enter whatever pattern you like.

closed 2018-03-30 by ccrisan

tigernero79, 2018-03-30:
as you said, it works, I put "% d-% m-% Y /% H-% M-% S" and it only goes to start now to get date and time on two lines with the timestamp that character to use?

tigernero79, 2018-03-31:
Solved "\n"