motioneye-project/motioneyeos #1462: Question: External Cameras

Dewey3, 2018-03-30:
I have been using MotionEyeOS with the Waveshare cameras, and very happy with it.  I would like to add a couple outdoor cameras to my setup.  In my searching "IP Cameras" here, I have read a response that said that MotionEyeOS was more designed to work with a single network connection.  Therefore, my question is can anyone give a suggestion on some decent external (outdoor type) cameras that I can use with MotionEyeOS running on my Raspberry Pi 3B that will overcome the 15 foot USB cable limit?

gizzburn, 2018-03-30:
I just bought two Iegeek ip cameras and they are working great with a R Pi 3, one is a 720P and the other is a 1020P . Both work well with motioneyeos but do need some tweaking to get the motion detection to work well.
Once setup you only need a single cable to power the camera and you can make this as long as you need, I extended both power leads so I could place the cameras in the best position.

https://www.amazon.co.uk/ieGeek-Waterproof-Security-Surveillance-Detection-Black-720P/dp/B073GPKH4S/ref=pd_rhf_gw_p_img_8?_encoding=UTF8&psc=1&refRID=88NJ905PQ9DRQKDEKJR6

MJ.

mb39924, 2018-04-04:
@gizzburn interesting camera. Is there a big difference between the 720 and 1080 version?

mb39924, 2018-04-15:
@gizzburn I bought the 720p camera because I was just too curious about the quality of such a cheap outdoor cam. Camera and image is way better than I guessed when using the CamHi app.

I tried adding this camer in my motioneyeos system but can't find  way to do it..
The only stream that seems to be working is a low quality jpg stream

How did you add this camera to motioneyeos?

mb39924, 2018-11-07:
I've been using this camera in my setup (motioneyeos on a raspi 3 as a central hub with three camera's / 2 raspi cams and one iegeek720 + an external hd for storage) for six months now. No problems at all, Excellent camera with great night vision. I've mounted the camera outside under the gutter and a wired internet connection.

About the weather: I've had rain and wind. no snow or ice yet. Only issue is  motiondetection when it rains (nothing to do with the camera itself). Oh, and spiders: webmaking is also triggering my motiondetection a lot.

jasaw, 2018-11-08:
@mb39924 You may be able to reduce the false motion detection from rain and spider web by using `despeckle_filter EeEedDdDl` option. The default `despeckle_filter` value in motionEyeOS is `EedDl`, which is great for most people, but depending on camera type, you may need to make it more aggressive by eroding twice. See http://emit.demon.co.uk/motion/

mb39924, 2018-11-12:
@jasaw Thanks for your reply. I shall give it a try :)

nickfoxall, 2018-11-29:
> I've been using this camera in my setup (motioneyeos on a raspi 3 as a central hub with three camera's / 2 raspi cams and one iegeek720 + an external hd for storage) for six months now. No problems [...]

What address do you use to connect to the ieGeek camera, took the plunge and ordered one but I just whatever address I seem to use to add it to motioneye I just get a single picture it seems in the preview area, motion eye picking up my local camera viewing that one ok just not the ieGeek!

mb39924, 2018-11-30:
Did you try this?

Add the camera as a network camera
use this URL: rtsp://xxx.xxx.x.xx/11
your ieGeek username and pw

I can post more detailed setting tomorrow if needed.

tedpugh, 2019-02-01:
Hi, I came across this post which is relevant to what I am doing. I have purchased an ieGeek network camera and cannot get it to work with motioneye. I can get it to recognise the camera but it is "unable to open video device". Can you post some more detailed settings install that may help me. For example, what is the user name? Is it the camera ID? and is the password the admin password? 
Any information gratefully received.

amitfalana, 2019-02-01:
Try and find in the below database. https://www.ispyconnect.com/sources.aspx

mb39924, 2019-02-01:
@tedpugh First thing you need to do is run the setup from the camera itself.  Wire connect the camera to your modem. There you set a username and password and if needed your wifi settings. I'm not at home so can't check but it is possible you have to enable rtsp from the camera settings. After you have done this you can add the camera to your motioneyeos system:

Add the camera as a network camera
use this URL: rtsp://xxx.xxx.x.xx/11
your ieGeek username and pw

mb39924, 2019-02-02:
I'm stil not at home but have you tried admin as the username?

tedpugh, 2019-02-03:
Thanks all for your help! The issue is now resolved. As far as I know, the camera was using UDP but MotioneyeOS was using TCP. The way to solve it (after many trials and much reading!) was to modify the **motion.conf** file in **/data/etc** and add the line **_rtsp_uses_tcp off_** . I could then add the camera using **_rtsp://xxx.xxx.x.x:554/onvif1_** with the camera ID and camera password and it then worked!

mb39924, 2019-02-04:
@tedpugh good to hear you've got the problem solved (and in an interesting way)
My iegeek camera just survived a lot of snow and ice. Great value for a small price.

tedpugh, 2019-02-21:
An update - for anyone interested. I found that if I added the camera as a UDP camera in the add camera menu it worked without mods anywhere else. I just did not see the drop-down option in the camera TCP/UDP....

SRowley90, 2019-05-31:
Sorry to reopen this discussion but I have recently got an ieGeek IE80 camera and want to link it into my MotionEyeOS setup. I've tried the suggestions above but I still get a "Connection Refused" error when entering my username and password in the MotionEye add camera console. I don't see an option in the settings to add a UDP camera. 

The IP for my camera is 192.68.1.37 so the URL I'm using is rtsp://192.168.1.37/11 and I'm entering my credentials for the app.

tedpugh, 2019-06-01:
No worries about reopening the thread! My ieGeek camera settings were:
rtsp://<cameraIP>:554/onvif1 - this might work for you. Also, selecting UDP is not intuitive... If I remember correctly you select the box which shows the TCP/IP option and then scroll down using the keyboard arrow key to show the UDP option. Took me a long time to find it. 
Let me know if it works for you. Ted

SRowley90, 2019-07-01:
I can't see any option to add a camera via TCP/IP never mind UDP. Think I'm resigning to the fact I can't see this [particular camera](https://www.amazon.co.uk/ieGeek-Security-Wireless-Surveillance-Tracking/dp/B07R5XY65M/ref=sr_1_7?crid=1UDUAC1TAU9Q2&keywords=iegeek+camera&qid=1561992575&s=gateway&sprefix=iegeek%2Caps%2C137&sr=8-7) within the MotionEyeOS system I have. 

[image]

crimperman, 2020-10-05:
> I can't see any option to add a camera via TCP/IP never mind UDP. Think I'm resigning to the fact I can't see this particular camera within the MotionEyeOS system I have.

Just in case anyone else ends up here. The UDP/TCP option appears only after you have authenticated/connected to the camera from motioneye. So first you must get rid of the Connection Refused message. To do this you must connect to the camera via rtsp.

* Add camera
* Select Network Camera
* Enter rtsp://xxx.xxx.xxx.xxx:554       <--- replace x's with your ip address. 
* Enter your camera's admin username and password
* Next to Camera (where it says Connection refused) you should see a drop down with the camera name and options for TCP/UDP

I'm not convinced it's needed though. My camera is listening on port 554 under tcp/rtsp. That said I can add the camera successfully but motioneye then complains it is unable to open video device. So I have some work to do :)

If you want to know how to connect to an iegeek camera via something other than the native app have a look here - it helped me https://www.ispyconnect.com/man.aspx?n=iegeek&page=0

starbasessd, 2020-10-05:
At least in current versions (20190911 and newer, to add any tcp/ip camera, you have several choices:
[image]
Network Camera (supports http:// and rtsp://
Remote Motioneye Camera (another motioneye on another computer (still using http:// or rtsp://)
Simple MPEG Camera (again, http:// or rtsp://)
I use Foscam cameras and d-link cameras, and use
Foscam:  rtsp://<ipaddress>:88/videoMain
d-link:     http://<ipaddress>:80/mjpeg.cgi
I use for RPiCams http://<ipaddress>:8081
All with proper usernames and passwords.