motioneye-project/motioneyeos #1463: no FastNetworkCamera: Pi Zero W + Logitech C310

fujinjr, 2018-03-31:
This is my setup:
Raspberry Pi Zero W + Logitech C310.

When I select FastNetworkCamera my system doesn't detect the Logitech C310.
Meanwhile when FNC is disabled it detect the C310 but with low FPS (as expected).

Any ideas? Thanks for your time.

ccrisan, 2018-03-31:
FNC only works with a Raspberry Pi CSI camera.

closed 2018-03-31 by ccrisan

fujinjr, 2018-03-31:
Is the Raspberry Pi Official Camera Module V2.1 8MP good?

ccrisan, 2018-03-31:
> Is the Raspberry Pi Official Camera Module V2.1 8MP good?

Should be good enough.

> Another question... when I activate Enable System Monitoring I see...

Hover them with the mouse and you'll get a tooltip of their meaning.

fujinjr, 2018-03-31:
It says streaming / capture framerate
So I assume: streaming is 'a', capture framerate is 'b', temperature is 'x', bandwidth is 'z', but...

```
a / b
x / y / z
```

**'y' ?**

ccrisan, 2018-03-31:
Yes, you are correct. After reading your question again, I realized it was about the system monitoring option on a RPi. Yes, it's temperature/sysload/netspeed.

A sysload of 2.5 on a 4 core CPU is fine. You should be worried when the average reaches 4.

fujinjr, 2018-03-31:
Isn't Raspberry Pi Zero W CPU's single core?

fujinjr, 2018-04-01:
@Moonbase are you also uploading movies to Google Drive?
It seems that USB camera causes more load for the system.