motioneye-project/motioneyeos #1464: howto capture high resolution images in interval capture mode

moritzthecat, 2018-03-31:
motionEye Version | 0.38.1
Motion Version | 4.1.1
OS Version | motionEyeOS 20180224
RPI Zero W with 5MP MMAL CIS camera 

I want to capture every xx seconds a full resolution 5MP image. But I can only capture HD (1920x1080) size images. Only in fast camera mode I can see the high resolution options of the camera. But using this mode NO interval capturing of images is possible. Have taken a look at the motion.conf settings but could not find an obvious tweak,

How might the interval capturing of high resolution images become possible ?

ecations, 2021-11-04:
I'm curious about this too.

starbasessd, 2021-11-04:
If you are doing both video and stills, remember the amount of processing needed to be done per frame of video for motion detection and the use of an ARM processor running relatively slow, along with the other things the OS side has to do, and a lack of RAM. On a PiZeroW, using a PiCam and the resolution above, I also get the slowness and lack of actual images. 
I adjusted Image Size to Custom, 2592x1936 (motion is very sensitive to images being modulo 16 sized) with 1 second intervals. It was able to process an image at about a rate of 1 per 6 seconds. Video on the WebGUI didn't change, but I was able to get still images to work (after a fashion). It is more than likely an issue with a single core slow ARM processor than anything else. It'd be interesting to test on a Pi3B+ or Pi4B-xGB to see if the processor & RAM make a difference, or if it is a motion issue... I don't have the PiCam to full sized ribbon cable, the the narrow PiCam to PiZero...