motioneye-project/motioneyeos #1467: Wifi not working on mt7601u on RPi zero W with internal WiFi broken

andreioniscu, 2018-04-02:
Hello, so I have a RPi Zero W with a broken internal WiFi module (I tried to solder something on the board in that area and I think I touched the WiFi controller). So I am now basically left with a non-W RPi Zero. My problem goes as this: I have found a mt7601u WiFi mico-adapter around the house, which by the way, works with Raspbian, but I can't get it to work in MotionEyeOS. The steps I took go like this:
1. Configure wpa_supplicant.conf in /data/etc with my SSID and PSK password.
2. Boot the RPi Zero for the first time after burning and modifying the SD card with my WiFi login info
3. Wait for everything to configure

At this point, I am greeted with the RPi Zero trying to start wpa_supplicant, but not succeding (Failed to start wpa_supplicant (mt7601u) ). After this error the RPi obviously reboots, because it has no network connection.

I figured it could be a driver issue with the mt7601u, but I can't even reach a command line, let alone build and install any drivers. On the other hand, the drivers should be precompiled right in to the linux kernel, at least from version 4.2 and up, from what I heard at least.

The thing is, this being a RPi Zero, there is no way to connect to ethernet without an adapter, which I'd rather not purchase just to get this set up, as my WiFi is pretty beefy and I really don't need any more speed for a surveillance system.

Any ideas? Thanks in advance!

ccrisan, 2018-04-02:
You could try to recover the log files by inserting the SD card into a Linux PC. You'll find them in the log directory of the data partition. `dmesg.log` and `boot.log` are of particular interest.

andreioniscu, 2018-04-02:
Ok, I have attahced boot.log dmesg.log and wpa_supplicant.log
Sorry for attaching them, I am new to github and I don't really know how to insert the text.
[boot.log](https://github.com/ccrisan/motioneyeos/files/1868146/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/1868147/dmesg.log)
[wpa_supplicant.log](https://github.com/ccrisan/motioneyeos/files/1868148/wpa_supplicant.log)

andreioniscu, 2018-04-02:
Ok, just wanted to make something clear, my SSID and my password do not contain quotes, but the log shows a pair of double quotes. what's that about?

ccrisan, 2018-04-02:
Can you please attach your wpa_supplicant.conf as well?

andreioniscu, 2018-04-02:
Sure, I have now modified it to delete the quotes, as it seems to already detect what's a password / SSID and what's not.
[wpa_supplicant.txt](https://github.com/ccrisan/motioneyeos/files/1868249/wpa_supplicant.txt)
(I renamed to txt because it wouldn't let me upload it)

ccrisan, 2018-04-02:
Well, those quotes are not optional. Please attach the wpa_supplicant that you actually use and that does not work.

andreioniscu, 2018-04-02:
Well, that's the one I actually use. Should I re-add the quotes?

andreioniscu, 2018-04-02:
Ok, so apparently I had a space between the first quote and the SSID and password. now that I fixed that and readded the quotes everything is working. Sorry for wasting your time and thanks for assisting me.

ccrisan, 2018-04-03:
Glad to hear you sorted it out.

closed 2018-04-03 by ccrisan