motioneye-project/motioneyeos #1468: ntpd not running

jasaw, 2018-04-03:
All my motionEyeOS cameras failed to sync time via ntp, all showing year 1970.

I had a brief power outage last night, and all my devices (motionEyeOS cameras, router, wifi APs and all other networking equipment in between) restarted at the same time when the power came back. The motionEyeOS cameras booted up faster than my router, and asked for DHCP before my router was ready, causing `ntpdate` to fail which is OK. MotionEyeOS should fallback to `ntpd` to sync time when the network comes back, but in this case `ntpd` stopped running quietly. `ps` command showed that `ntpd` process was not running. I didn't see any exit or error messages from `ntpd` in the logs. Any idea why ntpd process just quit silently? What about setting up a watchdog (like runinit) over essential services to restart them if they die?

This is the relevant section of `/var/log/messages`:
>Jan  1 10:00:52 meye-01 daemon.info dhclient: DHCPREQUEST on wlan0 to 255.255.255.255 port 67
Jan  1 10:01:15 meye-01 daemon.info dhclient: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 3
Jan  1 10:02:12 meye-01 user.notice wifi: disconnected
Jan  1 10:02:16 meye-01 daemon.info dhclient: No DHCPOFFERS received.
Jan  1 10:02:16 meye-01 daemon.info dhclient: Trying recorded lease 192.168.80.14
Jan  1 10:02:17 meye-01 user.notice wifi: disconnected
Jan  1 10:02:26 meye-01 daemon.info dhclient: No working leases in persistent database - sleeping.
Jan  1 10:02:27 meye-01 daemon.err ntpdate[1226]: name server cannot be used: Temporary failure in name resolution (-3)
Jan  1 10:02:27 meye-01 daemon.err ntpdate[1233]: name server cannot be used: Temporary failure in name resolution (-3)
Jan  1 10:02:27 meye-01 user.notice wifi: disconnected
Jan  1 10:02:27 meye-01 daemon.notice ntpd[1237]: ntpd 4.2.8p10@1.3728-o Thu Mar  1 05:07:32 UTC 2018 (1): Starting
Jan  1 10:02:27 meye-01 daemon.info ntpd[1237]: Command line: ntpd -g -c /data/etc/ntp.conf
Jan  1 10:02:28 meye-01 daemon.info ntpd[1242]: proto: precision = 3.000 usec (-18)
Jan  1 10:02:28 meye-01 daemon.info ntpd[1242]: Listen and drop on 0 v4wildcard 0.0.0.0:123
Jan  1 10:02:28 meye-01 daemon.info ntpd[1242]: Listen normally on 1 lo 127.0.0.1:123
Jan  1 10:02:28 meye-01 daemon.info ntpd[1242]: Listening on routing socket on fd #18 for interface updates
Jan  1 10:02:32 meye-01 user.notice wifi: disconnected
Jan  1 10:05:07 meye-01 daemon.info dhclient: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 6
Jan  1 10:05:07 meye-01 daemon.info dhclient: DHCPREQUEST on wlan0 to 255.255.255.255 port 67
Jan  1 10:05:07 meye-01 daemon.info dhclient: DHCPOFFER from 192.168.80.1
Jan  1 10:05:08 meye-01 daemon.info dhclient: DHCPACK from 192.168.80.1
Jan  1 10:05:08 meye-01 daemon.info dhclient: bound to 192.168.80.14 -- renewal in 35161 seconds.
Jan  1 10:05:10 meye-01 daemon.info ntpd[1242]: Listen normally on 2 wlan0 192.168.80.14:123

`boot.log`
>---- booting motionEyeOS 20180224 ----
> * Detecting disk device: /dev/mmcblk0
> * Mounting filesystems: done
> * Loading kernel modules: done
> * Setting hostname: done
> * Starting syslogd: done
> * Configuring CPU frequency: done
> * Starting throttle watcher: done
> * Starting eudev: done
> * Starting watchdog: done
> * Starting wpa_supplicant: done (brcmfmac)
> * Configuring wireless network: dhcp
>PING 192.168.80.1 (192.168.80.1): 56 data bytes
>
>--- 192.168.80.1 ping statistics ---
>1 packets transmitted, 0 packets received, 100% packet loss
> * Setting current date using ntp: failed
> * Starting ntpd: done
> * Starting crond: done
> * Starting sshd: done
> * Starting smbd: done
> * Starting nmbd: done
> * Starting motioneye: done
> * Executing user init script: Powering off HDMI
>done

ccrisan, 2018-04-03:
@jasaw have you explicitly set your date/time updating method to NTP? By default, HTTP is used.

jasaw, 2018-04-04:
@ccrisan yes, all my cameras are configured to sync time via ntp.

ccrisan, 2018-04-04:
Alright, that sounds like a bug then. I'll look into it.

ccrisan, 2018-04-20:
@jasaw I cannot seem to be able to reproduce this issue. The `ntpd` daemon starts no matter what the network state is. Can you reproduce it?

jasaw, 2018-04-23:
I spend little bit of time last weekend to produce it, but no luck, probably because it's timing sensitive. I need a more controlled environment to test this, which I haven't got time to set up.

jasaw, 2019-01-23:
My electricity went out again, and now all my cameras are facing this same problem. `ps` command shows ntpd not running, but /var/log/messages doesn't show anything useful.

Here's the relevant part of the /var/log/messages:
>Jan  1 10:02:29 meye-02 daemon.err ntpdate[1308]: name server cannot be used: Temporary failure in name resolution (-3)
Jan  1 10:02:29 meye-02 daemon.err ntpdate[1316]: name server cannot be used: Temporary failure in name resolution (-3)
Jan  1 10:02:29 meye-02 daemon.notice ntpd[1318]: ntpd 4.2.8p10@1.3728-o Thu Mar  1 05:07:32 UTC 2018 (1): Starting
Jan  1 10:02:29 meye-02 daemon.info ntpd[1318]: Command line: ntpd -g -c /data/etc/ntp.conf
Jan  1 10:02:29 meye-02 daemon.info ntpd[1321]: proto: precision = 3.000 usec (-18)
Jan  1 10:02:29 meye-02 daemon.info ntpd[1321]: Listen and drop on 0 v4wildcard 0.0.0.0:123
Jan  1 10:02:29 meye-02 daemon.info ntpd[1321]: Listen normally on 1 lo 127.0.0.1:123
Jan  1 10:02:29 meye-02 daemon.info ntpd[1321]: Listen normally on 2 wlan0 192.168.80.34:123
Jan  1 10:02:29 meye-02 daemon.info ntpd[1321]: Listening on routing socket on fd #19 for interface updates

When I start ntpd by hand, here's what I get:
>Jan  2 10:40:24 meye-02 daemon.notice ntpd[15800]: ntpd 4.2.8p10@1.3728-o Thu Mar  1 05:07:32 UTC 2018 (1): Starting
Jan  2 10:40:24 meye-02 daemon.info ntpd[15800]: Command line: ntpd -g -c /data/etc/ntp.conf
Jan  2 10:40:24 meye-02 daemon.info ntpd[15808]: proto: precision = 3.000 usec (-18)
Jan  2 10:40:24 meye-02 daemon.info ntpd[15808]: Listen and drop on 0 v4wildcard 0.0.0.0:123
Jan  2 10:40:24 meye-02 daemon.info ntpd[15808]: Listen normally on 1 lo 127.0.0.1:123
Jan  2 10:40:24 meye-02 daemon.info ntpd[15808]: Listen normally on 2 wlan0 192.168.80.34:123
Jan  2 10:40:24 meye-02 daemon.info ntpd[15808]: Listening on routing socket on fd #19 for interface updates
Jan 23 22:21:45 meye-02 daemon.info ntpd[15808]: receive: Unexpected origin timestamp 0x83abd980.c4f8defe does not match aorg 0000000000.00000000 from server@129.250.35.251 xmt 0xdff2ccc9.8bf4d06a

I've never seen `Unexpected origin timestamp` error message before, don't know if it's related.

Another thing I noticed is when my Internet connection came back, my ISP's DNS servers were giving resolve error for a few hours even though Internet was perfectly working. Don't know if it has anything to do with this problem.

ccrisan, 2019-01-23:
Could it be that `ntpd` maybe segfaulted at some point?

ccrisan, 2019-01-25:
@jasaw that's something we should probably do, indeed.