motioneye-project/motioneyeos #1470: monitor_1 script loses executable bit

rahammond, 2018-04-04:
I'm using MotionEyeOS version 20180224 on a Raspberry Pi 3 board, with an ELP brand USB camera model ELP-USB30W04MT-ML36.

I have a script _monitor_1_ located in /data/etc that contains the following to display the CPU temp on the camera window frame:

#!/bin/bash

cpuTemp0=$(cat /sys/class/thermal/thermal_zone0/temp)
cpuTemp1=$(($cpuTemp0/1000))
cpuTemp2=$(($cpuTemp0/100))
cpuTempM=$(($cpuTemp2 % $cpuTemp1))
echo CPU $cpuTemp1"."$cpuTempM" degC"
echo 10 1>&2

Once I make it executable, it works without issue, but if I make any changes in the web GUI and hit apply, it removes that script's ability to execute. Once I ssh in and run chmod +x /data/etc/monitor_1
it all works fine again. Any help you could provide would be greatly appreciated. It's quite likely I may be doing something wrong.

Great project BTW, I appreciate all of the time and effort that the author has put into this. A very feature-rich and easy to implement software.

[boot.log](https://github.com/ccrisan/motioneyeos/files/1876623/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/1876624/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/1876625/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/1876626/motioneye.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/1876638/dmesg.log)

ccrisan, 2018-04-04:
Is this 100% reproducible?

rahammond, 2018-04-04:
Yes, so far it has been.

ccrisan, 2018-04-04:
Can you enable debugging in expert settings, try reproducing the issue and attach the motioneye.log file again?

rahammond, 2018-04-04:
Once I enabled debug and did the reboot, I had to make monitor_1 executable again.
At that point I went back into the settings, enabled and disabled movies, and hit apply.
After that, monitor_1 was no longer executable.

Roger

[motioneye_debug_on.log](https://github.com/ccrisan/motioneyeos/files/1877455/motioneye_debug_on.log)

ccrisan, 2018-04-16:
@rahammond I found the problem. It appears to be a bug with the Raspberry PI-specific versions of motionEyeOS, where the "internal" monitoring script setting would interfere with the user-provided one.

More precisely, function [_set_sys_mon()](https://github.com/ccrisan/motioneyeos/blob/master/board/raspberrypi/motioneye-modules/boardctl.py#L205) removes the executable flag if the internal monitoring setting is disabled. A workaround would be to enable it (in Expert Settings) and provide your own script. It shouldn't mess with it afterwards.