motioneye-project/motioneyeos #1472: No boot on RPi Zero W

joelwetzel, 2018-04-06:
I'm trying to boot MotionEyeOS on a Raspberry Pi Zero W.  I'm using the latest release (20180401), and when I plug in power, the activity light blinks once, but then nothing else happens.  I have been able to get MotionEyeOS running on my Raspberry Pi 3B+, and I can run Raspbian on the Zero.  However, MotionEyeOS does not seem to boot on the Zero.  I've tried re-flashing multiple times.  (I'm just doing a basic flash of the .img onto a freshly formatted card.)  For the Zero, do I need to do a more involved installation?  Thanks!

joelwetzel, 2018-04-06:
Update:  It does boot on the Zero W if I go back to the 20180224 release.  So perhaps it's an issue with the latest release on the Zero W.

ccrisan, 2018-04-07:
20180401 has one one build which is for the RPi3 (for now). For the Zero, use 20180224.

closed 2018-04-07 by ccrisan

justlostintime, 2019-04-10:
I am running the latest version of motioneyeos for the pi 3 on pi zero w , it flashes one and just dies, move it to the three it runs sets up then just works. Move it back to pi zerow and it flashes once and just stops?

justlostintime, 2019-04-10:
This is a problem even with the 20180224 version as well! Is there a plan to have the os working on the Pi zero

ccrisan, 2019-04-12:
@justlostintime please run the Pi (1) version on the Zero W. The Pi 3 version is only for the Pi 3, as indicated in [Supported Devices](https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices)