motioneye-project/motioneyeos #1473: Live stream issue with Safari

altamij, 2018-04-06:
I setup motioneyeos on a raspberry pi 3, and it's been working fine for a couple years as a network streaming camera. However, now I am having issues accessing the live stream on Mac 10.3.4 and iOS 11.3. It loads the live stream, but only as a single picture. It does not continue to stream. If i open the link in chrome, it functions as intended (streaming video). Anyone else see this issue?

greenie1987, 2018-04-08:
I’m having the same issue on iOS 11.3. MotionEye was working great now on my iPhone running iOS 11.3 Safari only loads a still image and Chrome for iOS loads a screen that says “Unable to download file”. Not sure what’s happening? 

Chrome on my desktop is working as expected as with other browsers. Not sure what to do?

careyer, 2018-04-18:
Same issue here.... Since iOS 11.3 the stream does only load the first frame and displays a single frame. MotionEyeOS has been working for years without problems before. Seems like Apple changed something.

Maybe this can be a hint to start with:
https://github.com/webrtc/samples/issues/929

Seems like streams are no longer auto started at load and need aditional properties to do so.

ccrisan, 2018-04-18:
@careyer motionEye doesn't normally use the "stream". Instead it refreshes the image using JavaScript, to overcome all MJPEG browser restrictions. What type of camera have you added to your motionEye? How did you add it? Is it, by any chance, a Simple MJPEG Camera type?

careyer, 2018-04-18:
Hi Calin,

I am using a Standard RaspiCam (v1)... It has ever worked flawlessly and I did not touch motioneyeos at all. The Live-Video displaying in the browser broke when updating iOS to 11.3 as the previous posts also point to.

Cheers!

Dangermouse999, 2018-04-18:
I have exactly the same issue. Everything was working fine until the iOS11.3 update, now 1 frame is displayed and then things go wrong (Safari just stops, iOS Chrome reports a download error; Chrome on a Win7 PC works fine). Interestingly it’s only the actual camera streams that now have a problem. The camera preview panes on the admin page still seem fine. Could this be a clue/solution ?

jasaw, 2018-04-18:
This may be a motion issue. https://github.com/Motion-Project/motion/issues/700

careyer, 2018-04-18:
Possible but unlikely since neither motioneyeos nor the motion component of it was updated or changed in any way. I was the update to iOS11.3 that "broke" it ;-)

altamij, 2018-04-18:
I joined the beta program and reported the bug to apple as well (still exists in 11.4 so far)

ccrisan, 2018-04-18:
@careyer do you have Fast Network Camera enabled? If yes, if you try to disable it, does the problem persist?

careyer, 2018-04-18:
Hi Calin,

see my post above: "I am using a Standard RaspiCam (v1)...It **is NOT used in Fast Network Camera mode** because I need Motion detection." :-)

Cheers! Keep up the excellent work!

ccrisan, 2018-04-18:
@careyer that doesn't make any sense to me. I'm really sorry but I don't know how to help you. motionEye's code simply requests image after image and replaces the <img> tag src. Is there any way you could see if there are any javascript errors on your platform?

 If anyone can help us find the problem here and come up with a fix, I'd be happy to merge it into motionEye.

Dangermouse999, 2018-04-18:
If Apple won’t reverse/correct their change and nothing else seems to help, could an alternative way of presenting the streams perhaps be added? Maybe an additional port number for each camera could represent that image using the method employed on the admin screen? I know this is extra work and the problem doesn’t lie in MotioneyeOS but it’s so useful to have this working it would be an extremely useful workaround.

careyer, 2018-04-19:
@ccrisan : No there is no error message and i do not know of a method how to debug webpages on Safari for iOS. Since a lot of people have this problem (see posts above) I think this is clearly a bug induced with the latest iOS11.3 update, I can verify that the live video feed in the admin panel does work though.

ccrisan, 2018-04-19:
@careyer 
> I can verify that the live video feed in the admin panel does work though.

Well what doesn't work then? The motion MJPEG stream (that's not really used with motionEye)?

Dangermouse999, 2018-04-19:
I only use the mjpeg streams when accessing over the internet (it’s reasonably safe for port forwarding; one-way stream with no control inputs etc). It does seem that’s what has recently broken.

Thanks and best regards - Paul.

ccrisan, 2018-04-19:
Although I have no means to test this, I believe it's safe to say that recent Safari on iOS has some problems rendering MJPEG streams. I've seen some related issues on the motion project and I know that they only deal with MJPEG streams when it comes to presenting the stream to the user.

mszel42, 2018-04-24:
just as @jasaw commented, this is an issue with motion, more specifically about some recent change in Safari. If you go and get the embed url instead of the port then it's going to work as expected:
http://host/picture/1/frame/

altamij, 2018-05-01:
I reported the issue to apple and it appears that it has been fixed on their end in the latest beta. issue can be closed

closed 2018-05-01 by altamij

starbasessd, 2022-10-07:
In motionEyeOS I believe it is in Python 2.7, not javascript. Most, if not all the python code is located in 
/usr/lib/python2.7/site-packages/motioneye/ but they are .pyc files (precompiled).
You would need the ['source code'](https://github.com/motioneye-project/motioneye/releases) and use a utility like [uncompyle6](https://pypi.python.org/pypi/uncompyle6), then look for your keywords.