motioneye-project/motioneyeos #1475: No loop recording?

Cubytus, 2018-04-07:
Hello there, 

I looked into the documentation, but it doesn't appear this feature is available as of latest MotionEyeOS version today.

Sure, MotionEyeOS does allow for setting the number of days a recording should be kept, but it lacks a critical one: erase older recordings when disk space is below X%. In fact, there's no point in the files to be kept for a month, when the storage cannot contain more than a week's worth of recordings, or less if there are many events.

**Crude test setup**: I intentionally left the RPi in a rather passing area, set to cut movies in 2-minutes chunks so as to get individual files. It uses regular RPi NoIR camera. File storage set to /dev/mmcblk0p3, root directory set as /home/ftp/Camera1 (incorrect path as files are recorded in /data/ftp/Camera1; /home/ftp contains folders _storage_ and _sdcard_, the last one containing an empty _Camera1_ folder, last edited when MotionEyeOS was imaged to the SD card.

**Result**: The SD card filled in a few days; worse, the latest recorded date was 3 days ago. In other words, in a real-life situation, it would have missed events that may have happened yesterday.

**Proposed solution**: Such a basic functionadd a default-on setting "Erase oldest recording when disk is XX% full", another one to get an email alert when disk space goes below YY% so as to give user some time to backup important files or let him/her erase whatever files he/she finds unimportant.

Optionally: provide a way for the user to mark some files as "important, do not delete automatically" so the system would erase the next-oldest file and keep on recording events as they happen.

jasaw, 2018-04-09:
Look at issue https://github.com/ccrisan/motioneyeos/issues/1164 and https://github.com/ccrisan/motioneyeos/issues/785.

jasaw, 2018-04-18:
Duplicate. Closing issue.

closed 2018-04-18 by jasaw