motioneye-project/motioneyeos #1476: h264/omx bad quality when low frame rate

rdb9879, 2018-04-07:
I have a Motioneyeos setup on a raspberry pi zero W with the fast network camera mode enabled. I then have a Raspberry Pi 3 B connected to the Network camera, recording continuously using h.264/omx file settings, 100% quality. This was working good with 800x600, 10 FPS. In order to reduce the power consumption of the fast network camera, I kept everything the same except lowered to 5 FPS. After doing this, I no longer get quality looking videos: Everything is blocky or blurry, basically looks like a poor quality setting.

I used SSH to check the CPU utilization using "top", and the CPU is barely getting throttled (it's always under 10%, usually right around 5%). I can see a good image quality when I log into the RP3 web stream. I switched over the AVI file format, and now I am once again getting good quality videos at 5 FPS, although at a larger file size. The AVI file format has an extra annoyance where I get about 5 seconds of "frozen" video at the start of each file. Most of the other file formats couldn't find a player on my Windows machine.

Is there anything I can do to get h.264/omx back to good quality at 5 FPS?

jasaw, 2018-04-09:
@rdb9879 I've seen really blocky videos from h264_omx when I was implementing that feature, but that was because I wasn't setting the bitrate of the encoder. I have never seen this issue since then. How many network cameras is your Pi3 supporting?

Can you please attach your Pi3 thread-1.conf file and motion.log file? Also try playing around with the video quality, set to 99%, 75%, 25% on the Pi3, see if there's any difference.

jasaw, 2019-07-01:
Might be the same problem as https://github.com/ccrisan/motioneyeos/issues/1900