motioneye-project/motioneyeos #1479: Static IP

Croonci, 2018-04-09:
Hi,

I'm using a Raspberry Pi 3 B+ and the 20180401 release.
I can't enable the DHCP Server on my router, so I tried to connect with the static_ip.conf:
I formatted it as Unix (LF) file and copied it to the sd-card prior the first boot. 

It just says the following:

static_ip="192.168.1.32/24"
static_gw="192.168.1.1"
static_dns="192.168.1.1"

When I start the Raspberry, it says:
/data/etc/static_ip.conf: line 1: static_ip: command not found
/data/etc/static_ip.conf: line 2: static_gw: command not found
/data/etc/static_ip.conf: line 3: static_dns: command not found

What am I doing wrong? Thanks in advance for any help.

ccrisan, 2018-04-09:
Please attach your full boot log.

Croonci, 2018-04-09:
Hey,

I can't explain why but the issue is solved. I wrote the image another time on the SD card and copied exactly the same static_ip.conf i used before. It seems to work now. Strange because I rewrote the image like 2-3 times before and it didn't work.

I've got some problems with the Wifi connection now but will try to figure this out myself before bothering you anymore.

Thanks.

closed 2018-04-09 by Croonci