motioneye-project/motioneyeos #1480: RPi 3B+ camera not detected

gmccauley, 2018-04-10:
Just got another Pi 3B+ today and figured I'd give this a shot.  Installed pre-release 20180401 of MotionEyeOS, but my official Noir v2.1 is not being detected.  I know the camera and cable is good, because it works on another Pi running OctoPi.

Tried running raspistill and get the below:
[root@meye1 ~]# raspistill -k
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

Any suggestions?

ccrisan, 2018-04-10:
1. What does `vcgencmd get_mem gpu` say?
2. Does it work with Raspbian on the same setup?

gmccauley, 2018-04-11:
[root@meye1 ~]# vcgencmd get_mem gpu
gpu=256M

I did not try raspian yet.

rodo2020, 2018-04-14:
I have the same error with Pi 3 B.

ccrisan, 2018-05-03:
@rodo2020 
1. Have you tweaked anything or it simply doesn't work at all with the default motionEyeOS settings?
2. Are you sure you have plugged it in the correct socket?
3. Can you try Raspbian on the very same setup and see if the camera actually works?