motioneye-project/motioneyeos #1483: Possible to add camera from jpg via URL source?

smoke007, 2018-04-11:
Is it possible to add a "camera" that uses a URL as the source to display a jpg image?  Here's an example of the URL I'd like to use as the image to feed the "camera".  https://mysite.duckdns.org:8123/api/camera_proxy/camera.doorbell_pressed?api_password=mypassword

My goal would be for this camera to either refresh the image from the URL a couple times a minute or when the image changes.  I'm using motioneye as the mobile phone viewer for my cameras.  

I'm using Home Assistant for home automation.  I'm monitoring my doorbell button pushes with a Wemos D1 mini, so that HA takes a picture from my camera of who's at the door.  This jpg image is then stored at my sample URL above and publicly available over the internet.  HA then sends a push notification to my phone with that image and a link to my motioneye camera viewer page.  On that page, I can see the live view of the cameras, but would also like to see the static image that was taken when someone pushed my doorbell button as a "camera".

I tried every combination of adding this URL to motioneye as a camera, but no luck.

Thank you for your time!
Smoke.007

ccrisan, 2018-04-11:
See the Snapshot URL in the Video Streaming section.

smoke007, 2018-04-11:
Hi Calin,
I believe you're referring to Snapshot URL found under the camera's Video Streaming section for a camera that is setup.  That would get a JPEG out of the camera.  I'm trying to use https://mysite.duckdns.org:8123/local/snapshots/doorbell.jpg as the source for a camera feed.  (This is an example URL)  I've tried adding a Network Camera with both of my URL examples above.  motionEye recognizes the URL as a JPEG Network Camera type, but I only get a grey box with message, "Unable to open video device since".  The URL opens correctly in a browser and I can view the JPEG from both my local network where motionEye is and from the internet.
Thank you

ccrisan, 2018-04-11:
@smoke007 in that case, please look through motion.log and see if you can spot anything suspicious. If motion is not capable of using the JPEG as a stream, then either there's a problem with your JPEG when accessed by motion or there's a bug in motion.

In any case, I believe there's nothing motionEye can do about it. See [Reporting A Motion Issue](https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue) if you want to take it to the motion project.

closed 2018-04-11 by ccrisan

smoke007, 2018-04-12:
@ccrisan Thank you for the tip. Within the motion.log file, get the following message after trying to add a camera from **https**://mysite.duckdns.org:8123/local/snapshots/doorbell.jpg
I notice the mention of HTTP service, but my URL is **HTTPS**.  Is https URL not supported as feed?  It detects the camera type as "Network Camera" with this URL.

I searched for this error on the motioneyeos and motion before posting, but didn't find anything about my "Invalid netcam service 'http'" error below.

[1:ml1] [CRT] [NET] netcam_start: Invalid netcam service 'http' 
[1:ml1] [ERR] [VID] vid_start: Netcam failed to open
[1:ml1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[1:ml1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[1:ml1] [CRT] [NET] netcam_start: Invalid netcam service 'http' 
[1:ml1] [ERR] [VID] vid_start: Netcam failed to open
[1:ml1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[1:ml1] [CRT] [NET] netcam_start: Invalid netcam service 'http' 
[1:ml1] [ERR] [VID] vid_start: Netcam failed to open
[1:ml1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[1:ml1] [CRT] [NET] netcam_start: Invalid netcam service 'http' 
[1:ml1] [ERR] [VID] vid_start: Netcam failed to open
[1:ml1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera

Thank you in advance!