motioneye-project/motioneyeos #1489: External (powered) USB drive not listed in File Storage dropdown

enterprisemike, 2018-04-14:
Hi, me again :|

After my previous external USB issue (low voltage), I purchased and externally powered USB drive. When I first connected it everything was fine.
When I checked a month later, the USB is not listing, and storage had reverted to the internal SD card (which was almost full). 
dmesg.log seems to indicate that it's finding the drive OK, but it's still not listing in the interface.
[Motioneye.zip](https://github.com/ccrisan/motioneyeos/files/1909926/Motioneye.zip)

Thanks
Mike

jasaw, 2018-04-16:
From the logs, the hard drive was detected successfully, but sda1 was mounted and in use for quite some time, then changed to sda2. Did you do something to the hard drive or reconfigured the system?

Detected hard drive as sda with sda1 and sda2 partitions:
>[Thu Jan  1 10:00:03 1970] scsi 0:0:0:0: Direct-Access     Seagate  Expansion Desk   0911 PQ: 0 ANSI: 6
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] 5860533167 512-byte logical blocks: (3.00 TB/2.73 TiB)
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] 4096-byte physical blocks
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] Write Protect is off
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] Mode Sense: 47 00 00 08
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[Thu Jan  1 10:00:03 1970]  sda: sda1 sda2
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] Attached SCSI disk

Initially, this hard drive was in use:
>Jan  1 00:00:10 BSLS-Monitor daemon.notice ntfs-3g[919]: Mounted /dev/sda1 (Read-Write, label "", NTFS 1.2)
Jan  1 00:00:10 BSLS-Monitor daemon.notice ntfs-3g[919]: Mount options: allow_other,nonempty,relatime,fsname=/dev/sda1,blkdev,blksize=4096

Then changed to this:
>Jan  1 00:00:04 BSLS-Monitor daemon.notice ntfs-3g[777]: Mounted /dev/sda1 (Read-Write, label "SEA_DISK", NTFS 3.1)
Jan  1 00:00:04 BSLS-Monitor daemon.notice ntfs-3g[777]: Mount options: allow_other,nonempty,relatime,fsname=/dev/sda1,blkdev,blksize=4096

Then to this:
>Jan  1 00:00:04 BSLS-Monitor daemon.notice ntfs-3g[690]: Mounted /dev/sda1 (Read-Write, label "iOmega 1TB", NTFS 3.1)
Jan  1 00:00:04 BSLS-Monitor daemon.notice ntfs-3g[690]: Mount options: allow_other,nonempty,relatime,fsname=/dev/sda1,blkdev,blksize=4096

Finally to this on sda2 (this is sda2, different from the previous few hard drives that appeared as sda1):
>Jan  1 10:00:14 BSLS-Monitor daemon.notice ntfs-3g[807]: Mounted /dev/sda2 (Read-Write, label "Seagate Expansion Drive", NTFS 3.1)
Jan  1 10:00:14 BSLS-Monitor daemon.notice ntfs-3g[807]: Mount options: allow_other,nonempty,relatime,fsname=/dev/sda2,blkdev,blksize=4096

Now motioneye is unable to find sda1 because it's not there anymore. It's sda2.
>1970-01-01 10:01:19: [motioneye]    ERROR: could not find mounted partition for device "/dev/sda1" and target dir "/data/media/sda1/footage/StationNorth"
1970-01-01 10:01:21: [motioneye]    ERROR: could not find mounted partition for device "/dev/sda1" and target dir "/data/media/sda1/footage/ParkSouth"

enterprisemike, 2018-04-16:
HI,

Thanks for your help. 

The references to iOmega are from before I got it all working with the Seagate.

The Seagate was working at 2018-03-10, so I presume that was on sda1. When I checked again on 2018-04-14, it’s mounting on sda2 (from what you’re saying). There was no change made between those 2 dates (expect perhaps a restart that I’m not aware of).

If I understand correctly, there is a mount point at sda1, which motioneye is trying to read, but the Seagate is mounting on sda2. Is that correct? So to correct this I need to get the Seagate to mount on sda1. Is that right?

Thanks
Mike

From: jasaw
Sent: Monday, 16 April 2018 11:01 AM
To: ccrisan/motioneyeos
Cc: enterprisemike; Author
Subject: Re: [ccrisan/motioneyeos] External (powered) USB drive not listed inFile Storage dropdown (#1489)

From the logs, the hard drive was detected successfully, but sda1 was mounted and in use for quite some time, then changed to sda2. Did you do something to the hard drive or reconfigured the system?
Detected hard drive as sda with sda1 and sda2 partitions:
[Thu Jan 1 10:00:03 1970] scsi 0:0:0:0: Direct-Access Seagate Expansion Desk 0911 PQ: 0 ANSI: 6
[Thu Jan 1 10:00:03 1970] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[Thu Jan 1 10:00:03 1970] sd 0:0:0:0: [sda] 5860533167 512-byte logical blocks: (3.00 TB/2.73 TiB)
[Thu Jan 1 10:00:03 1970] sd 0:0:0:0: [sda] 4096-byte physical blocks
[Thu Jan 1 10:00:03 1970] sd 0:0:0:0: [sda] Write Protect is off
[Thu Jan 1 10:00:03 1970] sd 0:0:0:0: [sda] Mode Sense: 47 00 00 08
[Thu Jan 1 10:00:03 1970] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[Thu Jan 1 10:00:03 1970] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[Thu Jan 1 10:00:03 1970] sda: sda1 sda2
[Thu Jan 1 10:00:03 1970] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[Thu Jan 1 10:00:03 1970] sd 0:0:0:0: [sda] Attached SCSI disk
Initially, this hard drive was in use:
Jan 1 00:00:10 BSLS-Monitor daemon.notice ntfs-3g[919]: Mounted /dev/sda1 (Read-Write, label "", NTFS 1.2)
Jan 1 00:00:10 BSLS-Monitor daemon.notice ntfs-3g[919]: Mount options: allow_other,nonempty,relatime,fsname=/dev/sda1,blkdev,blksize=4096
Then changed to this:
Jan 1 00:00:04 BSLS-Monitor daemon.notice ntfs-3g[777]: Mounted /dev/sda1 (Read-Write, label "SEA_DISK", NTFS 3.1)
Jan 1 00:00:04 BSLS-Monitor daemon.notice ntfs-3g[777]: Mount options: allow_other,nonempty,relatime,fsname=/dev/sda1,blkdev,blksize=4096
Then to this:
Jan 1 00:00:04 BSLS-Monitor daemon.notice ntfs-3g[690]: Mounted /dev/sda1 (Read-Write, label "iOmega 1TB", NTFS 3.1)
Jan 1 00:00:04 BSLS-Monitor daemon.notice ntfs-3g[690]: Mount options: allow_other,nonempty,relatime,fsname=/dev/sda1,blkdev,blksize=4096
Finally to this on sda2 (this is sda2, different from the previous few hard drives that appeared as sda1):
Jan 1 10:00:14 BSLS-Monitor daemon.notice ntfs-3g[807]: Mounted /dev/sda2 (Read-Write, label "Seagate Expansion Drive", NTFS 3.1)
Jan 1 10:00:14 BSLS-Monitor daemon.notice ntfs-3g[807]: Mount options: allow_other,nonempty,relatime,fsname=/dev/sda2,blkdev,blksize=4096
Now motioneye is unable to find sda1 because it's not there anymore. It's sda2.
1970-01-01 10:01:19: [motioneye] ERROR: could not find mounted partition for device "/dev/sda1" and target dir "/data/media/sda1/footage/StationNorth"
1970-01-01 10:01:21: [motioneye] ERROR: could not find mounted partition for device "/dev/sda1" and target dir "/data/media/sda1/footage/ParkSouth"

jasaw, 2018-04-16:
I'm not sure how your hard drive is partitioned. From dmesg, looks like your hard drive has 2 partitions (sda1 and sda2):
>[Thu Jan  1 10:00:03 1970] scsi 0:0:0:0: Direct-Access     Seagate  Expansion Desk   0911 PQ: 0 ANSI: 6
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] 5860533167 512-byte logical blocks: (3.00 TB/2.73 TiB)
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] 4096-byte physical blocks
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] Write Protect is off
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] Mode Sense: 47 00 00 08
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[Thu Jan  1 10:00:03 1970]  sda: sda1 sda2
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[Thu Jan  1 10:00:03 1970] sd 0:0:0:0: [sda] Attached SCSI disk