motioneye-project/motioneyeos #1491: Attempting to Configure Motion Detection Settings as Admin Deletes Camera and Causes Error

abhinans, 2018-04-15:
Hello,

I have tried all of the latest releases of motioneyeOS for raspberrypi 3 (02/24, 04/01 (pre) and 04/15 (pre)). When I flash the image onto my 2GB SD card along with a wpa_supplicant.conf file. I am able to access the GUI as expected. 

But when I login as admin and configure motion detection along with an email smtp setup and apply the changes. The system complains of an error and recommends a refresh. On refreshing, I am logged out. I don't see the camera anymore (I'm using raspberry pi cam 2) and it says a camera has not been added. Further, the motion detection settings are no longer visible in the GUI. 

I have tried this after re-flashing several times as well as deleting my browser cache, but the issue persists. Please let me know how I can fix this issue.

abhinans, 2018-04-15:
So I did some further investigation and it happens only when I try to setup the notification email.

ccrisan, 2018-04-16:
Please try recovering a `motioneye.log` file from a broken setup and attach it here.

abhinans, 2018-04-17:
I have attached my motioneye.log file. Thanks!
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/1917955/motioneye.log)

ccrisan, 2018-04-20:
@abhinans what do your email settings look like? They are the ones responsible of breaking your configs.

abhinans, 2018-04-20:
I'm using tls, port 587, smtp.gmail.com and my password has the special character ' 

Is that the issue? Please let me know if you need more info.

abhinans, 2018-04-20:
It was the special character ' causing the issue. I changed my password and it worked. Thanks! I'm closing the thread.

closed 2018-04-20 by abhinans