motioneye-project/motioneyeos #1492: No Fast Network Camera option with RPi2 and Raspicam

Cornmeister, 2018-04-16:
Hi,

I'm running multiple instances of MotionEyeOS, all on Raspberry Pi models 2.
first one is running the following version, and does have FNC enabled.
Camera type : Simple MJPEG Camera
motionEye Version | 0.38.1
Motion Version | 4.1.1
OS Version | motionEyeOS 20180224

The second Pi has the same hardware, but has no FNC option, 
also the camera type looks different.
Camera device: vc.ril.camera
Camera type: MMAL Camera
motionEye Version | 0.39
Motion Version | 4.1.1
OS Version | motionEyeOS dev2018041

any advise how to get the FNC enabled on the second pi ?

ccrisan, 2018-04-16:
@Cornmeister whered did you download motionEyeOS version "dev2018041" from? Can you please point me to the link?

Cornmeister, 2018-04-17:
its just your nightly-dev
https://github.com/ccrisan/motioneyeos/releases/tag/nightly-dev
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi2-dev20180415.img.xz

ccrisan, 2018-04-17:
Then I assume OS version `dev2018041`  is a typo and it's in fact `dev20180415`?

Cornmeister, 2018-04-17:
damn, that was a typo indeed :/

ccrisan, 2018-04-17:
In that case, thanks for testing the nightly builds. In fact I'm working on putting together a nightly build system (which is incredibly difficult given the required build time for BuildRoot).

The building system is now using some caching mechanism and therefore these are not clean builds. I need to further investigate and see what's going on with the missing feature.

Cornmeister, 2018-04-17:
I'll try the stable for now, will try the nightly builds later on. tnx for building this awesome images!

ccrisan, 2018-04-20:
Fixed in dev20180419 via https://github.com/ccrisan/motioneyeos/commit/14a84ada8feac5866d3cca725bb27ab4778608f9.

Thanks for reporting the issue.

closed 2018-04-20 by ccrisan