motioneye-project/motioneyeos #1493: can't set up network share on Pi 3B

ianporter13, 2018-04-19:
got a strange issue where i cant connect a network share on a new install on a new Pi 3 B+

I'm running motionEyeOS 20180401 and mirrored the settings I have on my Pi Zero W that works fine

I can connect to the share from anywhere else but it wont connect from this new device

currently using FTP to the same share as a workaround

any ideas?

ccrisan, 2018-04-20:
Please give us some more details. What error do you get? Can we see a log file?

jka3588, 2018-04-21:
I'm having the same issue, but am fairly new to github. What can I provide that may help? The error that occurs when I try and add a network share is "Accessing network share failed: cannot mount network share!"

ianporter13, 2018-04-21:
Yes that’s the error I see
I’ll get the logs off over the weekend

It had totally crashed this morning for some reason too 🙁

So far for diagnostics I’d followed a thread with a similar issue where you’d given details on trying to mount the drive from SSH but that doesn’t work either

jka3588, 2018-04-21:
Check and see if you have a similar message to this, because that's what I'm seeing

SMB2.1) specify vers=1.0 on mount.
[Wed Dec 31 19:04:28 1969] CIFS VFS: cifs_mount failed w/return code = -22
[Wed Dec 31 19:04:28 1969] No dialect specified on mount. Default has changed to a more secure dialect, SMB2.1 or later (e.g. SMB3), from CIFS (SMB1). To use the less secure SMB1 dialect to access old servers which do not support SMB3 (or SMB2.1) specify vers=1.0 on mount.
[Wed Dec 31 19:04:28 1969] CIFS VFS: cifs_mount failed w/return code = -22

ianporter13, 2018-04-23:
here's the log files from my installs
the one from the Pi 3B+ is the one with the issue
the one from the Zero is connected to the same share and works fine

[motioneye (working Pi Zero).log](https://github.com/ccrisan/motioneyeos/files/1938853/motioneye.working.Pi.Zero.log)

[motioneye (not working Pi 3B plus).log](https://github.com/ccrisan/motioneyeos/files/1938858/motioneye.not.working.Pi.3B.plus.log)

ccrisan, 2018-06-25:
This should be fixed via https://github.com/ccrisan/motioneye/commit/1bce4dd358774f69e52fa3814eeea54e9fc6b156. It will be part of the next prerelease.

closed 2018-06-25 by ccrisan

ccrisan, 2018-06-26:
@ianporter13 can you please try the latest [nightly dev build](https://github.com/ccrisan/motioneyeos/releases/tag/nightly-dev) and see if the problem has been fixed?

ianporter13, 2018-06-26:
Won’t be able to test for a few weeks as away from home

Ian

Will-PH, 2018-06-27:
Just tested on my end and it seems to be working again, thanks for the fix.

[image]