motioneye-project/motioneyeos #1495: Support for Banana Pi PI-ZERO

careyer, 2018-04-19:
It would be awesome to see support for this board.
Same form-factor as Raspberry Pi Zero but Quad-Core and thus very well suited for acceptable framerates with motion detection.

http://www.banana-pi.org/bpi-zero.html

Thanks!
Keep up the excellent work!

ccrisan, 2018-04-19:
https://github.com/ccrisan/motioneyeos/wiki/FAQ#can-motioneyeos-be-ported-to-insert-your-board-here

freeflocw, 2018-05-15:
I would send you a banana pi zero (quadcore)  from germany or more easy should i transfer some money  so you order direct to have/test one? 
Thank you for your excellent work. Best serious and usefull Raspi project.

freeflocw, 2018-05-15:
Even It looks like normal rpi cam module do not work with bananaPiZero 
https://forum.armbian.com/topic/4801-banana-pi-zero/?page=1

ccrisan, 2018-05-15:
@freeflocw please contact me via email at ccrisan at gmail.

freeflocw, 2018-05-15:
Hello Calin,

Would you like to try to test MEos on Banana Pi zero? 
Which camera module?

I can order it here and send it to you directly:
https://m.reichelt.de/Einplatinen-Computer/BANANA-PI-ZERO/3/index.html?ACTION=3&LA=446&ARTICLE=218297&GROUPID=8242&artnr=BANANA+PI+ZERO&SEARCH=banana%2Bpi%2Bzero&trstct=pos_0

But no cameramodule!

There should be also nano and orange pi zero with H2 H3 . 

I just tried the Wifiaccess works excellent , 
but i have a problem there is no ssh (enabled disabled) on a pi3 . Is this normal?
Normaly , no wifiaccess installation it works. I will try other boards. 

Do you get enough donation for your eccelent work?
 I use it very long, no more chiniese phoning-home cloudcams. 

Beste Gruesse aus Muenchen
Florian Biehler
+49 1792960218