motioneye-project/motioneyeos #1496: Support Orange PI Zero (WIFI MODULE)

ZEROCL, 2018-04-19:
Hello! I downloaded the .IMG of the OrangePI One to mount it in my Orange PI Zero, and I see many fewer configurable options than those granted by MotionEyesOS for Raspberry PI for example, I do not see how to back the photos and videos to dropbox, or googledrive, and neither for the WIFI, because of course, the Orange PI One does not have WIFI, but it works by network cable in the Orange PI Zero, my doubt is as follows, it is possible to enable the WIFI option to detect the WIFI that if it has integrated the Orange PI ZERO?

Currently I managed to extend the range of options in terms of the configuration of the OrangePI Zero, acquired the options of resplaint in the Cloud, etc, everything works except the WIFI. How did I do it? I installed MOtionEyesOS in raspberry PI, once operative I made a backup, then this backup is imported from the Orange PI Zero with the MotionEyeSO for Orange Pi One.

Until then I managed to solve it, but what I need is to leave the WIFI that the Orange PI Zero brings to work.

Please help!

Sorry for my English, I write from Chile.

ccrisan, 2018-04-20:
> I do not see how to back the photos and videos to dropbox, or googledrive

Can I see a screenshot?

> ...and neither for the WIFI, because of course, the Orange PI One does not have WIFI

motionEyeOS doesn't work that way; you should see the WiFi network options regardless of the hardware you're on; again, screenshots, please.

ZEROCL, 2018-04-20:
Thanks for answering!

1.- Excuse me, I made a mistake when describing the situation regarding file backup in googledrive and dropbox, this works great! It works perfect!

2.- Regarding this, Yes I can see the WIFI setting! and to put a fixed IP, but it does not work the option to connect to WIFI, it is visible, but it is not operating, checking by command line it is possible to see that there is no Wi-Fi card detected, therefore I guess it would be enough to add it but for lack of permits and drivers (I guess) I can not do it.

I will attach screenshots below, however, I would like to describe again that the image I used is from OrangePI One, I mounted it in my Orange PI Zero, but as the options without any more reduced than with the Raspberry PI, what I did (a experiment mode) was from the raspberry PI with the MotionEyeOS system for Raspberry, was to create a backup and then export this same backup to the OrangePI Zero. (We must remember that the OrangePI Zero I left running with the MotionEyeOS system for OrangePI One)

Then after mounting the backup of the raspberry in the Orange PI Zero! Magic! I can see all the options! the support in the cloud if it works wonders! in googledrive and dropbox, the only thing that does not work is the operation of the WIFI, can I add the WIFI in command line? For lack of superuser permission I can not do it.

I hope I have been clearer! and thanks!
Now go the screenshots!
[image]
[image]

ZEROCL, 2018-04-20:
As I said, the WIFI option appears functional, but it is not operative, it does not connect to the WIFI, I assume that the driver of the WIFI card is not installed.

Running this on the orangePI zero will significantly reduce costs. Hopefully we can solve all this and achieve progress.

Thank you!

ccrisan, 2018-04-22:
After carefully reading the issue again, I now realized it's about Orange Pi Zero. Well, Orange Pi Zero is not supported by motionEyeOS and the fact that the Orange Pi One image partially works on the Zero is due to various similarities between the two boards.

I don't own an Orange Pi Zero board so if you want me to try to make it work on such a board, you can [send me one](https://github.com/ccrisan/motioneyeos/wiki/FAQ#can-motioneyeos-be-ported-to-insert-your-board-here) and I'll see what I can do.

closed 2018-04-22 by ccrisan

ZEROCL, 2018-04-22:
To be able to help you without problem, I could send you an Orange PI Zero but the issue of the shipping cost is the complex.
However to help you if you want help me to install the WIFI Driver in the MotionEyesOS system, I need necessary access to install it.

I hope you can understand me.

Thank you!

ccrisan, 2018-04-23:
You have the source code here. You have the [building instructions](https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source). I'm happy to review your pull request.

logiota, 2020-02-27:
I can send you one, but please be quick since China is tricky now.

starbasessd, 2020-02-27:
Just curious, but have you tried a standard supported USB wifi dongle, like
EDIMax?
https://www.amazon.com/Edimax-EW-7811Un-150Mbps-Raspberry-Supports/dp/B003MTTJOY
THere are several other models listed on sites like adafruit.com...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?