motioneye-project/motioneyeos #1497: How to change video output

eatalotofpie, 2018-04-20:
Hi!
I was tasked to build a 'super simple and idiot proof CCTV system by my boss two days ago. I decided to use motioneyeos and so far everything has been good.

I have on issue though: 
I'd like to change the video output so that the video feed from the single webcam connected to the Raspberry Pi 3 (model B+ if that makes a difference) would show up on the TV connected via HDMI to the Pi. This would allow us employees to see the front door from the office through the aforementioned TV monitor.

Is this possible? All help appreciated.

jasaw, 2018-04-20:
There are a few ways of doing this:
1. Set up another device, Raspberry Pi or whatever that has HDMI to your TV. Open a web browser on this device and point it to your motionEyeOS Raspberry Pi.
2. This option is more complicated. Rather than using motionEyeOS, you use Raspbian with desktop, install [motionEye](https://github.com/ccrisan/motioneye/wiki/Installation), and configure it. When you're happy with the configuration, set Chromium web browser to start automatically in kiosk mode at boot time, and point your Chromium to localhost. It should show you motionEye's web interface.

ccrisan, 2018-04-20:
@eatalotofpie if you're using the Raspberry PI CSI camera, you could enable HDMI preview.