motioneye-project/motioneyeos #1503: motioneye as AP

ioTataryn, 2018-04-30:
Hello,

I have tried to follow the following doc without success:

The pi zero W says it cannot connect to wired network then reboots.

I have used etcher to flash the latest image, and rather than copying over wpa_supplicant i have copied over hostapd.conf, and dnsmasq.conf, as well trying with os.conf (all on first boot)
when i attach wpa_supplicant on the first boot it work fine.
 
hostapd:

 ssid=camera
 wpa_passphrase=camera1234
 wpa=2
 wpa_key_mgmt=WPA-PSK
 interface=wlan0
 channel=1
 driver=nl80211
 hw_mode=g
 ieee80211n=1
 ieee80211ac=1

dnsmasq:

interface=wlan0
dhcp-range=192.168.27.50,192.168.27.150,24h
dhcp-leasefile=/data/dnsmasq.leases

os.conf:

os_debug="false"
os_prereleases="false"
os_tty_login="tty1"
os_eth="eth0"
os_wlan="wlan0"
os_ppp="ppp0"
os_networkless="true"
os_country="GB"
os_firmware_method="github"
os_firmware_repo="ccrisan/motioneyeos"
os_firmware_username=""
os_firmware_password=""

anyone know what i am doing wrong? I am copying the 2 or 3 files to the root of the newly created filesystem on the sd card then booting the pi Zero W.  It always gets stuck on the wired network, then reboots.  I have tried os_networkless set to true and false also.  What am I misisng?

Thanks!

ccrisan, 2018-04-30:
Logs? What are you trying to achieve? And also there's an issue template that we expect users to fill out when submitting a new issue. That way now I wouldn't have to ask for things like version and logs.

OttoWinter, 2018-05-07:
Seeing this as well using `motioneyeos-raspberrypi-20180401.img.xz` on a Raspberry Pi Zero W.

I'm trying to set up a simple motioneye webcam (awesome project, btw!) with an Access Point as there's no WiFi network I can easily connect to at the location I'm setting this up.

When setting up, the configured WiFi AP does show up in the WiFi lists on my MacBook and Pixel XL. So the WiFi SSID broadcast are apparently sent. The problem is that when I try to connect to the AP with any device I've tested, they say some error happened and they can't connect to it.

I haven't set up a monitor yet so there might be something in the logs I'm missing. Another possibility might be that somehow dnsmasq doesn't hand out IP addresses and that's what's causing the error.

Anyway, as this issue shows up using the base AP configuration from the wiki, I think it should be fixed or there should be a note in the docs about configuration.

**Steps to reproduce:**

 1. Use a Raspberry Pi Zero W (maybe also happens with other devices, haven't tested yet) with the `motioneyeos-raspberrypi-20180401` image.
 2. Flash the image to a new SD card and copy these files to `/boot/`:

**`hostapd.conf`**:
```
ssid=myssid
wpa_passphrase=12345678
wpa=2
wpa_key_mgmt=WPA-PSK
interface=wlan0
channel=1
driver=nl80211
hw_mode=g
ieee80211n=1
ieee80211ac=1
```

**`dnsmasq.conf`**:
```
interface=wlan0
dhcp-range=192.168.27.50,192.168.27.150,24h
dhcp-leasefile=/data/dnsmasq.leases
```

**`os.conf`** (the issue also happens without this file):
```
os_networkless="false"
```

 3. Eject the drive, plug it into the RPi and start it up.
 4. Observe that after a while the WiFi AP `myssid` shows up as expected. Optionally wait some time to make sure the setup is complete.
 5. Try to connect to the AP. Fails with my devices.

I've successfully set up `hostapd` APs like this before on debian. Next thing I'll try is to get to the logs somehow (I don't have a mini HDMI cable...) and try different dnsmasq configurations.

Thanks!

ccrisan, 2018-05-07:
It's most likely due to dnsmasq not functioning properly. I'll have to try and reproduce this myself, but in the meantime could you please:
1. Recover some log files from your unit (like boot.log, messages) and look for dnsmasq-related issues.
2. Try the stable 20180224 version and see if the problem is still there.