motioneye-project/motioneyeos #1504: wget command not found

CheetahGod, 2018-05-01:
While trying to connect motion eye os running on my raspberry pi to google drive the wget command is not found.

ccrisan, 2018-05-01:
@CheetahGod, this issue is invalid for the following reasons:

 * you have totally ignored the issue template we have in place
 * you have not read our [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ#can-i-install-some-package) which clearly says that this is not a regular OS where you can install stuff

Long story short: use `curl`.

closed 2018-05-01 by ccrisan

geronimo1400, 2018-06-10:
@ccrisan The confusion here is because your directions for connecting to Google Drive in your FAQ says to use wget.

miaosun, 2018-06-13:
yes true @ccrisan , in the FAQ says to use wget, you should update it

jasaw, 2018-06-14:
@geronimo1400 & @miaosun The Google Drive FAQ is obsolete, says so in bold on top of the page.

duncangroenewald, 2019-04-10:
Better to replace the instructions with instructions for how to sync with Google drive then.  I can't seem to find any instruction for how to set that up.