motioneye-project/motioneyeos #1505: Yolo classifier + AI - stick ( movidius, laceli )

MrJBSwe, 2018-05-01:
I think combining motion detection with classification like [yolo](https://pjreddie.com/darknet/yolo/) and using AI-sticks as workhorse could fit very nicely in to the core purpose of motioneyeos.

The trend is a usb-dongle drawing less than 1 watt  could do the classifying job, example [movidius](https://www.pyimagesearch.com/2018/02/19/real-time-object-detection-on-the-raspberry-pi-with-the-movidius-ncs/) or [laceli](https://www.cnx-software.com/2018/01/03/laceli-ai-compute-stick-is-a-more-efficient-alternative-to-intel-movidius-neural-compute-stick/)

jasaw, 2018-05-02:
This is very interesting. I've been looking for ways to add some sort of AI to enhance motion detection. I found someone using the [Pi GPU](https://petewarden.com/2014/08/07/how-to-optimize-raspberry-pi-code-using-its-gpu/).
Do you know roughly what kind of performance we can expect running Yolo classifier on a RPi 3?
Also do you know whether there's a pre-trained model that we can use or whether there's readily available training data? The performance of the AI depends heavily on the training data, and seems like the most difficult part to get right if we were to train it ourselves.

MrJBSwe, 2018-05-02:
>Pi GPU
Based on [JeVois](https://www.youtube.com/watch?v=d5CfljT5kec), my guess is PI-GPU & tiny Yolo will run at 0.5 - 2 fps

>pre-trained
based on [coco](http://cocodataset.org/#home)
wget https://pjreddie.com/media/files/yolov2-tiny-voc.weights

Movement => Classify
[Tiny yolov2 uses 7.1 GFlops](https://pjreddie.com/darknet/yolov2/) which makes it a good starting point for PI-GPU & movidius. It can also be run on CPU. It seems quite "easy" to train [skynet for squirrels](https://hackaday.io/project/156926-squirrel-deterrent), [fish](https://youtu.be/UytgeEKM1SY) 

Continuous classification
With [laceli](https://www.youtube.com/watch?v=VOC3huqHrss), I think movement detection might be obsolete, since [yolo is very robust](https://www.youtube.com/watch?v=VOC3huqHrss) to light changes and noise from background. [Yolo v3](https://www.youtube.com/watch?v=s8Ui_kV9dhw)  uses 140 GFlops and my guess is it should run at 5-10 fps on laceli.

wb666greene, 2018-05-20:
The sample code from here:
[https://www.pyimagesearch.com/2018/02/19/real-time-object-detection-on-the-raspberry-pi-with-the-movidius-ncs/](url)

which uses MobileNet SSD (I believe from here: https://github.com/chuanqi305/MobileNet-SSD) and a PiCamera module to do about 5 fps on a Pi 3 with Movidius NCS.

I've modified the PyImageSearch sample code to get images via MQTT instead of the PiCamera video stream and then  run object detection on them.  If a "person" is detected I write out the detected file which will ultimately get pushed to my cell phone in a way yet TBD.

I've written a simple node-red flow also running on the Pi3 with the NCS that presents an ftp server and sends the image files to the NCS detection script. The Pi3 also runs the MQTT server and node-red.

I then configured motioneyeOS on a PiZeroW to ftp its motion images to the Pi3 node-red ftp server.

Its working great, been running all afternoon.  Since virtually all security DVRs and netcams can ftp their detected images, I thing this system has great generality and could produce a system worthy of a high priority push notification since the false positive rate will be near zero.

I plan to put it up on github soon, but it will be my first github project attempt so it might take me longer than I'd like.

Running the "Fast Netcam" or v4l2 MJPEG streams into the neural network instead of "snapshots" might be even better, but the FLIR Lorex security DVR I have uses proprietary protocols so ftp'd snapshots is what I used.  There is a lot of ugly code to support the lameness of my DVR so after I got it working (been running for three days now) I simplified things for this simple test system I plan to share as a starting point project to integrate AI with video motion detection to greatly lower the false alarm rate.

To suggest an enhancement to motioneye, I'd like to see an option for it to push jpegs directoy to an MQTT server instead of ftp.

I don't think video motion detection like motioneye is obsolete, it makes a great front-end to reduce the load on the network and AI subsystem letting more cameras be handled with less hardware.

debsahu, 2018-05-29:
@wb666greene using ocv3 causes fps to drop below 1 (~0.88). Using yolov2 with native PiCamera library is a struggle, I tried. Getting started with GitHub is not straight forward, was a struggle initially. My suggestion is use GitHub desktop.

MrJBSwe, 2018-05-31:
Security-cam example
https://ai.intel.com/practical-applications-of-deep-learning-build-a-diy-smart-security-camera-using-the-intel-movidius-neural-compute-stick/

wb666greene, 2018-06-05:
Thanks for the extra background info.  I'll add some of these links to my github, think they are very helpful and more enlightening that anything I could write up.

My main point is that  I've made an "add-on" for an existing video security DVR instead of making a security camera with AI.  Expect these to flood the market soon, it'll be a good thing!  but until then, I wanted AI in my existing system without a lot of expense or rewiring.

MotioneyeOS is a a perfectly good way get a simple video security DVR going, and in fact it has far superior video motion detection than does my FLIR/Lorex system, but you are on your own for "weatherproofing" and adding IR illumination to your MotioneyeOS setup -- not a small job!

I used it so I could give a simple example instead of over-complicating things with all the ugly code needed to deal with my FLIR/Lorex system's lameness.

MrJBSwe, 2018-07-30:
Living on the edge
https://aiyprojects.withgoogle.com/edge-tpu

MrJBSwe, 2018-08-13:
dlib has very interesting functionality for face recognition
https://medium.com/@ageitgey/machine-learning-is-fun-part-4-modern-face-recognition-with-deep-learning-c3cffc121d78

http://blog.dlib.net/2017/02/high-quality-face-recognition-with-deep.html

That could run on pi...
https://www.pyimagesearch.com/2018/06/25/raspberry-pi-face-recognition/
( But to do a serious job acceleration with movidius is necessary )

this is quite impressing
https://github.com/srianant/computer_vision/blob/master/output/face_rec.gif

jasaw, 2018-10-18:
I just had a play with Movidius on a RPi 3B+ recently, with version 2 of the NCSDK (still in beta). Here's what I've found:
- NCSDK v2 works after fixing a few installation scripts. It pulls in a lot of dependencies, and wants very specific versions of libraries. It will most likely break other existing programs on your host machine by messing with the libraries. It is recommended to install on an sacrificial machine.
- It supports several neural net models, but neural net models are pulled in from external repositories, so things are left in a broken state because of changes on those external repositories.
- I tried various neural net models, and only found one that is reliably detecting person.
  - YOLO: unable to test, does not seem to be supported.
  - TinyYOLO: a fast neural net model, but very low accuracy, completely useless for our application.
  - GoogleNet: I can't find pretrained weights that's geared towards detecting person.
  - SSD MobileNet with default pretrained weights: does not seem to be detecting person.
  - SSD MobileNet with chuanqi's pretrained weights: does a good job at detecting person, dogs and cats are OK, struggles with the rest. Still very good for our application. https://github.com/chuanqi305/MobileNet-SSD
- Frame rates: I'm getting around 6.5 fps, far from real-time.
- Resolution: Before pushing an image into the Movidius NCS, the image needs to be scaled down to the specific resolution that was used to train the neural net model, which is 300x300 for chuanqi's model.

With all that said, Movidius can still be used for our application as a 2nd-pass system that combs through all the recorded videos to detect person in non-real-time. This may be useful for various use cases, for example:
1. Send notification and/or trigger alarm when a person is detected. No need to manually look through recorded videos anymore.
2. Remove videos and/or images that do not contain a person. This reduces storage requirements.

On 2nd thought, Movidius can still be used for real-time person detection. Integrate into motion software and feed one frame every 3rd or 4th frame into the Movidius. I haven't seen an NCSDKv2 C API, but surely it has to exist right?

wb666greene, 2018-10-18:
Thanks for posting this. I was about to try and install ncsdk V2 on an old machine to give tiny yolo a try, saved me a lot of time wasting!  I was hoping the increased resolution of the TinyYolo model over MobileNetSSD  would help.  Any possibility you could upload or send me your compiled graph?  I'd still like to play with the model, but you've removed my motivation for getting setup to compile it with the SDK.

Have you tried any of the multi-threaded or multi-processing examples?  Running on my i7 desktop I've found using USB3 only improved the frame rate by less than half a frame/sec over USB2.

I've been running Chuanqi's MobileNetSSD since July on a Pi3B handling D1 "snapshots"  from 9 cameras with overlapping fields of view from my pre-existing Lorex security system.  I use the activation of PIR motion sensors to filter (or "gate" ) the images sent to the AI to reduce the load.  It works great, I get the snapshots via FTP, and filter what goes to the AI.  Only real complaint is the latency to detection can be as high as 3 or 4 seconds, although usually its about 2 seconds, other than the latency it seems real-time enough for me -- effectively the same as motioneye 1 frame/second snapshots.

Your use case (1) was my goal.  I never looked at the video anyways, as the Lorex system's "scrubbing" is so poor.  With the Emailed AI snapshots I now have a timestamp to use should I ever need to go back and look at the 24/7 video record (what the Lorex is really good at, but everything built on top of it is just plain pitiful).  

I have three system modes, Idle, Audio, and Notify.  Idle means we are home and going in and out and don't want to be nagged by the AI.  Audio means were are home, but want audio notification of a person in the monitored areas -- fantastic for mail and package deliveries.  Notify sends Email images to our cell phones.  The key is the Audio AI mode has never woken us up in the middle of the night with a false alarm, and the only Emails have all been valid, mailman, political canvasser, package delivery, etc.

Much as I like Motioneye and MotioneyeOS I'm finding the PiCamera modules are not really suitable for 24/7 use as after a period of several days to a couple of weeks the module "crashes" and only returns a static image from before the crash.  Everything else seems to work fine SSH, node-red dashboard, cron, etc. but the AI is effectively blind until a reboot.  I've a software only MobileNetSSD AI running on a Pi2B and Pi3B with Pi NoIR camera modules, while it only gets one AI frame about every 2 seconds, it still can be surprisingly useful for monitoring key entry areas, but the "soft" camera failures is a serious issue.  I've not ever run Motioneye OS 24/7 long enough to know if it suffers the issue or not.  I should probably setup my PiZeroW and try.

With this experience, I' starting to swap out some Lorex cameras with 720p Onviif "netcams"  (USAVision, ~$20 on Amazon) since I don't really care about the video, its a step up in snapshot resolution (1280x720), one Pi3B+ and Movidius can handle about four cameras with ~1 second worst case detection latency.

In 24/7 testing usage I am getting Movidius "TIMEOUT" errors every three or four days.  It seems I can recover with a try block around the NCS API function calls, and having the except deallocate the graph and close the device, followed by a repeat device scan, open and load graph.  Tolerable amount of blind time once every few days.  I plan to rewrite for the V2 API to see if it fixes the issue, the V2 multistick example doesn't seem to have any errors yet in over a week of running.

wb666greene, 2018-10-19:
@jasaw 
I think I can confirm your comments about TinyYolo.  I got the sample code to run and modified it to input some D1 security camera images, detection performance is terrible.  Missing full frontal, full length, people in the center of the frame,  detecting shadows on the sides as "people", and all manner of wrong calls (two chairs as a bicycle, etc.)

Looks like MobileNetSSD is the only practical AI for security camera use at present on resource constrained systems.

While MobileNetSSD also makes a lot of wrong calls, if you only care about detecting "people", which seems fine for secruity camera systems, it performs very well in my experience.

MrJBSwe, 2018-10-21:
https://www.crowdsupply.com/up/ai-core-x

wb666greene, 2018-10-21:
@MrJBSwe 
I've seen it, but it looks like about $500 for the AI board and computer to plug it into (unless you already have one with a suitable interface, I don't).

Also so far the development environment looks like only C/C++ at present.  If python bindings become available for it I'll get a whole lot more interested.  Not that I'm any great python guru, but I find it really hard to beat for "rapid prototyping".

At this point I think the AI network is more of a limitation for security system purposes than is the hardware to run the network.  MobileNetSSD cpu only can get 7+ fps on my i7 desktop with the OpenCV 3.4.2 dnn module and simple non- threaded python code.

jasaw, 2018-10-25:
I have integrated(hacked) Movidius into motion and used ChuanQi's MobileNetSSD person detector to replace the old motion detection algorithm. If "Show Frame Changes" in motionEye is enabled, it will also draw a red box around the detected person and the confidence percentage at the top right corner. I have repurposed motion's "threshold" option to mean confidence threshold in percentage, which takes a range from 0 to 100 as integer. A detected person is only considered valid if the neural net confidence level is above this threshold.

This is very much a proof of concept, and only works with 1 camera stream, and only tested on a Raspberry Pi with Pi camera.

Code is here:
https://github.com/jasaw/motion/tree/movidius

How to use:
1. Install Movidius NCSDKv2. See https://movidius.github.io/ncsdk/install.html
2. Git clone the movidius branch into `/home/pi/motion` directory. Make sure you clone into `/home/pi/motion` directory because I have hard-coded the graph path for now.
3. Go into motion directory and run:
    - `autoreconf -fiv`
    - `./configure`
    - `make`
    - `sudo make install`
4. Before running `motion`, change the `threshold` in `thread-1.conf`. `75` seems like a good starting point.

wb666greene, 2018-10-25:
@jasaw 
This is very nice work.  Can I ask what kind of frame rate are you getting?  If you are getting significantly better fps than I am, I'd be motivated to re-write in C/C++

Using the V1 ncsapi and ChuanQi's MoblileNetSSD on a Pi3B I'm getting about 5.6 fps with the Pi camera module (1280x720) using simple python code and openCV 3.4.2 for image drawing, boxes, and labels (the python PiCamera library I use creates a capture thread).

With simple threaded python code I'm also getting about 5.7 fps from a 1280x720 Onvif netcam (the ~$20 one I mentioned in an earlier reply).  This same code and camera running on my i7 Desktop (heavily loaded) is getting about 8 fps.  On a lightly loaded AMD quad core its getting about 9 fps

Have you seen any performance improvements of V1 vs. V2 of the ncsapi?

I now have one Pi3B setup with the V2 ncsapi and have run some of the examples (using a USB camera at 640x480)  I was most interested in the multi-stick examples, but I've found that the two python examples from the appzoo that I've tried are in pretty bad shape -- not exiting and cleaning up threads properly.  I pretty much duplicate their 3-stick results but I don't think they are measuring the frame rate correctly.  The frame rate seems camera limited as dropping the light level drops the frame rate and their detection overlays show incredible "lag"

jasaw, 2018-10-25:
@wb666greene I don't know exactly how many frames I'm getting from the Movidius stick. I'm not even following the threaded example. I think it doesn't matter anyway as long as it's running at roughly 5 fps. With my implementation, everything still runs at whatever frame rate you set, say 30 fps, but inference is only done at 5 fps. A person usually doesn't move in and out of camera view within 200ms (5fps), so it's pretty safe to assume that we'll at least get a few frames of the person, which is more than enough for inference.

I'm going to refactor my code so that I can merge it into upstream motion, and have multi-stick support as well.

jasaw, 2018-10-26:
I have implemented proper MVNC support into motion software. See my earlier post for usage instructions: https://github.com/ccrisan/motioneyeos/issues/1505#issuecomment-432882027

MrJBSwe, 2018-11-06:
I have recently tried Nvidia Xavier => I get like 5fps 
https://devtalk.nvidia.com/default/topic/1042534/jetson-agx-xavier/yolo/

Since it is quite expensive, I'm still putting my hope in direction of AI - sticks like Movidius X

RK3399Pro is an interesting addition
https://www.indiegogo.com/projects/khadas-edge-rk3399pro-hackable-expandable-sbc#/

wb666greene, 2018-11-07:
@jasaw 
Interesting results on the thermal test,  I'm not seeing much fps difference between short duration tests (~10-30 seconds) and long test runs (overnight or longer).

I have given up on the v2 SDK for now, sticking with the V1 sdk and made some code variations to see what frame rates I can get with the same Python code (auto configuring for Python 3.5 vs 2.7) on three different systems comparing Thread and Multiprocessing to the baseline single main loop code which gave 3.2 fps for the Onvif cameras and 5.3 fps for a USB camera and openCV capture.  The Onvif cameras are 1280x720 and the USB camera was also set to 1280x720.

These tests suggested using three Python threads, one to round-robin sample the Onvif cameras, one to process the AI on the NCS, and the main thread to do everything else (MQTT for state and result reporting, saving images with  detections, displaying the live images, etc.) would be the way to go.

I got 10.7 fps on my i7 desktop with NCS on USB3 running Python 3.5  on an overnight run.  

Running the same code on Pi3B+ with Python 2.7 I'm getting 7.1 fps, but its only been running all morning.

My work area is ~27C and I'm not seeing any evidence of thermal throttling (or it hits so fast my test runs haven't been short enough to see it).   I don't think the v1 SDK has the temperature reporting features,  I haven't checked the "throttling state"  as I really only care about the "equilibrium" frame rate I can obtain.  For my purposes 6 fps will support 4 cameras.  I'm going to try adding a fourth thread to service a second NCS stick.

@MrJBSwe 
The Movidius NCS only costs ~$75, come on in the water is fine :) 
Running off a Pi3B+ its only using maybe 8W of power and your total entry fee is <$150 if you have the basics like spare keyboard, mouse, and monitor for installation and development.  My target environment is stand-alone headless networked "IOT" device talking to the outside world via MQTT, Email, Telegram etc.

MrJBSwe, 2018-11-07:
@wb666greene 
> come on in the water is fine
I have 2 movidus and like them as a "appetizer" while waiting for movidius x ( or something similar )
I have played around with both v1 &2 of ncsdk. I'm currently playing around on a [nivida 1070](https://octoperf.com/blog/2018/09/02/acer-predator-helios-500-mods/) to see what's possible when HW is less of a constraint. Yolov3 seems to be a bit overkill and drains even a 1070 of it's juice.

I want to run yolov2 ( or something similar ) at >= 4 fps ( yolo tiny gives too random results ). I plan to check out your code wb666greene & jasaw, interesting work !

this is a bit interesting ( price is right ;-)
https://youtu.be/bBuHOHPYY7k?t=69

jasaw, 2018-11-07:
@wb666greene From what I've read, thermal reporting is only available on v2 SDK. In my test, I'm pushing 11 fps consistently through the stick until it starts to go in and out of thermal throttle state cycle of 8 fps (thermal throttled) for 1 second, 11 fps (normal) for 3 seconds. If you take the average, it's still pushing 10 fps, which may explain the 10.7 fps that you're seeing on your i7 desktop. I imagine at higher ambient temperature like summer 45 degrees Celsius, it's going to stay throttled for much longer, possibly even go into 2nd stage throttle.

MrJBSwe, 2018-11-10:
Maybe something...   
https://www.96boards.ai/products/rock960/

Similar to Khadas Edge, RK3399Pro and the upcoming Rock Pi 4 & RockPro64-AI, I guess the trend is RK3399Pro for multiple reason ( but a I still hope for movidius X and or Laceli )

jasaw, 2018-11-15:
@MrJBSwe Yes, Neural Compute Stick 2 has finally been released. Let's see if I can get one to play with.

I see a few obstacles in supporting NCS 2.
- NCS 2 only works with Intel's new toolkit called OpenVINO.
- OpenVINO does **not** run on ARM machines, so will not run on Raspberry Pi.
- OpenVINO only provides C++ & Python API, but Motion software is written in C. I have a feeling that Motion developers are reluctant to switch to C++.
- Even if ARM is supported, cross compiling OpenVINO API looks like a giant pain.

Excited and disappointed at the same time...

wb666greene, 2018-11-15:
@MrJBSwe @jasaw 

Found the same issues jasaw lists above  with the NCS2 when going through the docs initiall.  I've ordered one form Mouser for $99+tax and shipping, they had 2000+ in stock when I ordered.

The OpenVINO won't install on Ubuntu 18.04, I tried on a fresh install I'd used with the NCS for testing/developiong multistick code, didn't get past the initial installing dependencies as at least one is unmet and the process stops.  So add fragile development environment to the list.

They say only 16.04 is supported.  I've made a fresh test install of 16.04 but the old drive I used seems to have errors so I didn't get very far but the it appeared to get past the dependencies installation,   Starting the installation GUI died with writing to /tmp errors and lots of disk I/O errors in dmesg.

I may have to re-install and start over with a different drive as the badblocks scan is taking forever and finding lots of errors :(

You can call C++ libraries from C, if the compilers match.  I've done it in a previous life, using a proprietary C++ library compiled for Linux from the device vendor.  It required a very specific  (old) version of gcc/g++ but I did get it to work.

Here is a brief overview of what you are up against:
https://www.teddy.ch/c++_library_in_c/

jasaw, 2018-11-15:
@MrJBSwe The IoT market is dominated by ARM, and Intel has been giving developers financial incentives to switch over to their Intel platform but without much success. By locking NCS2 down to Intel platform, I can see significant growth in Intel's IoT platform.

MrJBSwe, 2018-11-16:
@ jasaw 
I'll try to return them to mouser I really dislike what Intel has done here...
I guess they want to lose the "AI-stick" market to China....

I'll take a look at this
https://www.sophon.ai/post/36.html

MrJBSwe, 2018-11-16:
@wb666greene  "backdoors"
https://www.cs.vu.nl/~ast/intel/
( and on top of that Win10 ...like a ship full of holes )

Fair play and opensource...I guess we will have to wait for some kind of "FPGA" where we can flash our own hardware ...

wb666greene, 2018-11-20:
I've got the openVINO SDK installed on an old i3 running Ubuntu-Mate 16.04.  There were some path issues between the install doc on the website and where things ended up, but once I figured this out, all the tests passed and the demos I've tried all appear to run fine.  Looks like GPU support is broken for this old i915 motherboard, but the good news is it auto-detects Modivius vs Movidius2.

Not played with much yet, but with the interactive_facedetection_demo sample code and a USB WebCam the performance looks to be significantly better with the NCS2

NCS    facedetection: ~16 fps   face analysis: ~3 fps
NCS2                         ~42 fps                          ~10 fps
CPU                           ~17 fps                          ~5.4 fps
Note that the CPU needed an FP32 model where the NCS used fp16.   As with my MobleNet-SSD Python code, the CPU on the i3 is about the same as the Movidius NCS.

But I've an project using multiple NCS and MobileNet-SSD on where these old i3 systems would be the target (since I have them) so my first exercise will be to attempt to modify it to use the openVINO SDK and see if: 1) the original NCS does a bit better (library improvements) and 2) how much improvement I get with the NCS2  vs original NCS.  Next I'll investigate if perhaps this "face detection" works better than the "person detector" for my security system purposes -- an image with a face would be better when Emailed than an image of perhaps a person's back or an image with the head cut off :)

IMHO the Intel software support will blow away anything you will be able to get from those Chinese companies

To download the openVINO I had to "register" and they sent me an activation code that I was supposed to need.  But the installation never asked me for it, maybe I missed it and its why the GPU code doesn't work?  I've no real feel for the "power" of this old i3-i915 integrated graphics as a co-processor so its not a priority for me to pursue this.

FYI, Intel says they only support Ubuntu 16.04 and Windows 10,  I initially tried installing it on Ubuntu 18.04 which I had already installed on this old i3, and it didn't get very far (library version conflict)

MrJBSwe, 2018-11-20:
@wb666greene
Does NCS2 work with NCSDK 2.x ?  ( meaning, I could run it on an ARM )

>FYI, Intel says they only support Ubuntu 16.04
For the NCSDK 2.x the had put an version control in the script, it could actually run on 18.04
( maybe the same issue with openvino)

Have you tried Yolov2  ? ( not tiny )

What type of low energy board do you see as an realistic alternative to ARM based ?
Like the upcoming odroid-h2-intel-celeron
https://www.cnx-software.com/2018/10/19/odroid-h2-intel-celeron-j4150-development-board/

wb666greene, 2018-11-22:
I plan on trying my NCS2 with the v2 SDK I've installed on a Pi3B+,
unfortunately my router has acted up and I'm barely running.  With
Thanksgiving tomorrow it will be a while before I can get to it.

I've got some i3 "NUC" class boxes that with the V1 SDK and multi-threaded
Python are getting near linear speed up with multiple NCS (10.5 fps 1 NCS,
20 fps 2 NCS, 29fps 3 NCS,  4 NCS was starved for data) and seem to run the
openVINO pretty well although the i915 GPU support seems broken.  They are
not really suitable for battery power but they are less than 50W fully
loaded with a SATA hard drive and run off a single 12V DC supply.

See the comments section in this tutorial/blog:
https://www.pyimagesearch.com/2018/11/12/yolo-object-detection-with-opencv/
 With yolo v3 this i3 took about 1.5 seconds per image.  This v3 yolo
worked well on security camera images that original tiny yolo (available
for eh NCS) failed miserably on.

When I tried installing openVINO on 18.04 the initial  install
prerequisites failed with apt-get errors.  Not wanting to fight initially,
I installed 16.04 to a spare partition on the disk  and got some of the
demos running without too much hassle.   On this i3  with MobileNet-SSD CPU
only was about the same fps as with  the NCS.   Not the same network, but
on the USB webcam face detection openVINO demo code the CPU and NCS were
again about the same frame-rate (17fps vs 16fps) but plugging in the NCS2
it was significantly faster at 42 fps.

jasaw, 2018-12-04:
@MrJBSwe The Laceli based [Orange Pi AI stick](https://www.cnx-software.com/2018/11/22/orange-pi-ai-stick-2801-neural-compute-stick-sdk) does come with an SDK. If you click through to the [AliExpress](https://www.aliexpress.com/store/product/Orange-Pi-AI-Stick-2801-with-PLAI-Model-Transformation-Tools-Neural-Network-Computing-Stick-Artificial-Intelligence/1553371_32958159325.html) page, you'll see that they'll supply the SDK as well. I haven't seen any feedback on how good the SDK is, but at least it runs on ARM ! Cost of AI stick + SDK is prohibitive, at $218.

MrJBSwe, 2018-12-06:
@jasaw 
I have ordered one + sdk and I have also decided to keep my 2x NCS2 ( Intel seems to have a plan to support ARM ) and if not I have bought an Odroid H2.

wb666greene, 2018-12-20:
News flash, OpenVINO release 5 to support Raspberry Pi!

Just got a Email notification of its release.  Downloading it now.

Chiny91, 2018-12-21:
Santa visited Chiny Towers early and delivered a sacrificial R Pi 3b and a Movidius.   So, I have had the @jasaw recipe working with no problems for 12 hours and all appears well, although far too early to comment on reliability.   I'm sure the postman delivering today could not imagine the excitement he generated 😄 

Like many others, I'm plagued with unwanted motion triggers; cats, foxes, spiders, dawn, dusk, trees in wind, all of which have been eliminated by this AI system, so far.   Over the holiday period, I'll be looking at how motionEyeOS can deliver remote alerts, something I could not contemplate before, for fear of being overwhelmed.

I'd like to be able to contribute but I'm just an experienced user, occasionally dabbling in scripts, not a developer.   I did notice that at 1024x576 resolution, motionEye needed multiple restarts, but at 704x576, a single start is OK.   Whether that is relevant/general, I don't know how to ascertain.

MrJBSwe, 2018-12-30:
I've got yolov3 running on myriad ( NCS2 ) 1.7 fps  ( 3fps with CPU mode )
https://software.intel.com/en-us/articles/OpenVINO-Using-TensorFlow#inpage-nav-8
https://github.com/mystic123/tensorflow-yolo-v3
https://richardstechnotes.com/category/techniques/

Chiny91, 2019-02-19:
I've had my sacrificial R Pi 3b and a Movidius running for 2 months now, most successfully.   The @jasaw recipe has been 100% reliable, no mystery crashes or hangups, plus all running unattended whilst I was off site for 4 weeks.   On all but one day there were zero false positives; with 3 false positives being triggered in one hour on a single day by snow fall (a rare event round here).   There were zero missed events, based on reviewing a non-Movidius motionEyeOS cam covering a similar area.   To my Movidius Pi, I added some Python triggering a Pushover notification to my phone, giving me near real time notification of someone approaching my front door.

For my purposes (domestic security cams), this is a game changing development with false triggers being practically eliminated.   It does come at a cost of a Movidius but the benefits are well worth it for my use.   I do hope this development continues in the motion project and in due course, reaches motioneye and motionEyeOS.

jasaw, 2019-02-20:
@Chiny91 Thank you for doing extensive testing on it. I'm glad that the Movidius detection system has been working well for you. I would love to see motion officially support Movidius as well, but unfortunately Intel has decided to deprecate the NCSDK in preference of Intel's own OpenVINO framework. I haven't got time to look into getting motion to support OpenVINO yet, so there's still more work to be done.

MrJBSwe, 2019-03-05:
Google  Edge TPU 
https://coral.withgoogle.com/

wb666greene, 2019-03-06:
@MrJBSwe 
Thanks for the info, I thought I'd signed up for an Email when they became available, seems you can get them now from Mouser.  I would order one now, but I've got an Odroid 4UX arriving today which should be a worthwhile speedup over the Pi3B+ at about the same price, actually about 50% more when you factor in the price difference and the fact it needs about twice the power of the Pi3B+

After I've seen what benefits I get form this Odroid, I may order one of these edge dev boards, $150 seems pretty reasonable.

Although I'm using Caffe model MobileNetSSD, so moving to a tensorflow model will be an extra step

wb666greene, 2019-03-06:
@Chiny91 
I believe jasaw is using the same MobileNetSSD modle as I am.  I initially started using motioneyeos as a "frontend" to ftp images to my AI system, but my ultimate goal was to use images from our commercial security DVR which I've accomplished.  After running for almost a year I've had very few false alarms, most are in static locations and are typically plants and hanging baskets or clothes left on chairs, generally at night or as the sun rises or sets.

I'd be curious to see your snow induced false alarm images.  

We haven't had any snow here (rare, but had some last year before I had the system up) I'd wager snow in the top of plants would trigger the  false alarms similar as I got when we covered some plants in the camera field of view for a freeze warning.

As time goes by I'm developing a set of "spacial" filters to reject detections in certain x-y image areas of certain widths and heights.  With 15 cameras I've only needed about 8 rejection filters, most cameras have needed none, the camera that sees the pool deck has needed filters for most of the plants on the sides.  Cropping the image first, or masking it is another adhoc fix once I find out where these infrequent but statically located false detections are.  But since the AI gives me x-y location and object h&w this was very easy to do with a function where I pass in x,y,h,w of the detection with the x,y,h,w of the false detection and a tolerence, default 10%

MrJBSwe, 2019-03-16:
We Have a Lift-Off  !? ( looks very promising .. just got it running )
https://github.com/PINTO0309/TPU-MobilenetSSD

wb666greene, 2019-03-24:
@MrJBSwe 
The Coral TPU Accelerator USB stick is supported on most any Linux, I assume for the USB support compared to needing Windows drivers.  Its only a Python API at the moment.  I just got one, and getting the software installed and running their demos was about as easy as it gets.  I did it on my i7 Desktop with Python 3.5.2

 I've learned a lot form PINOT0309's code.

Now I've got more work to do to decide between the Coral or NCS2 for the next stage of my project.  The Coral was far easier to get going than was the NCS2, so unless its a dog performance wise I will likely go with it over the NCS 2 since they are competitive in price, if I recall the Coral was a bit less enpensive.

I just also got an Odroid UX4 Pi-like computer that runs Ubuntu-Mate 18.04 out of the box, it has USB3 and GigE.  Patches have just been posted to build OpenVINO on Ubuntu 18.04  so I hope to compare these two soon on this Pi-like processor after I get my NCS code up on GitHub.  

The code I plan to share supports multiple NCS sticks, can also use CPU AI threads in addition to or instead of the NCS (not useful on the PI), and also runs on Windows 7 and 10.  It works very well with MotioneyeOS or Motion ftping motion detection images to the AI for analysis.  Input images can be a mix of jpeg images via MQTT (which is fed by node-red ftp server) or "live" Onvif-type http web request jpeg snapshots or rtsp streams.  It is heavily threaded.

I would like to plant the seed of a request for a feature to be added to either Motioneye/OS or Motion (not sure where would be most efficient) to add an option to transfer motion images via MQTT instead of ftp.  I use node-red to implement an ftp-server and a localhost MQTT broker feeds them into my Python AI code.

Another Motioney/OS or Motion feature that would tie in nicely with sending motion detection images via MQTT would be to include the "box points" of the motion detection region like is an option to be drawn on the images.  This would be very useful to the AI especially with 1080p and above cameras since the AI is done on a 300x300 pixel resize, cameras better than 1080p are too much of a good thing.  Having the motion detection box points would make a good place to crop the image to feed to the AI instead of shrinking the entire frame.

I would be happy to help with the MQTT code (Python or C) if someone knows their way around Motion or Motioneye/OS well enough to integrate it.  Diving into Motion is way to much for me to bite off at the moment.

wb666greene, 2019-05-01:
> @Chiny91 Thank you for doing extensive testing on it. I'm glad that the Movidius detection system has been working well for you. I would love to see motion officially support Movidius as well, but [...]

@jasaw 
Turns out the latest OpenVINO makes it pretty trivial to modify CPU only OpenCV dnn module AI to use the NCS and or NCS2.

Details are here:
https://www.pyimagesearch.com/2019/04/08/openvino-opencv-and-movidius-ncs-on-the-raspberry-pi/#comment-515277

Since you seem to know your way around the Motion software very well, instead of integrating the AI into Motion, I think it'd be better to modify Motion (or MotioneyeOS) to have an extra option, similar to ftp motion snapshots, where instead of ftp it sends them to an MQTT broker and includes the "box points" of the motion detection region (corners of the red box overlaid on the images) as part of the message.  This way the AI could be a separate add-on to Motion or MotioneyeOS and "focus" on the region where the motion was detected.

I'm already doing this (minus the box points) with a node-red flow to implement the ftp server and send the snapshots to the AI via MQTT here:
https://github.com/wb666greene/AI_enhanced_video_security/blob/master/README.md
I've already modified this code to use OpenVINO, but NCS and OpenVINO code doesn't mix, so I haven't posted it yet. 

MQTT is much more efficient than ftp so doing it directly as an alternative option to ftp would be a win.

I can definitely help with getting MQTT working in either Python or C, I just don't know my way around Motion or MotioneyOS well enough to know where to start.

jasaw, 2019-05-02:
@wb666greene Great work ! Sending jpeg images to an AI sub-system is definitely a worthwhile add-on. I would go further and send recorded videos to the AI sub-system too, but feeding video frames to the neural net requires the video to be decoded first. How we transfer the images/videos to the AI sub-system can be independent of the AI sub-system. Adding MQTT as an option alongside FTP sounds like a good idea. If we were adding MQTT, I would add it to MotionEyeOS.

As for the "focus" box, I don't think that is necessary. From my testing so far, MobileNet SSD neural net seems to do a damn good job of recognizing humans without telling it where to focus. I would not apply an "AND" operation to the detection which increases false negatives. It's better to have false positives rather than false negatives to be on the safe side.

Just an update on my finding on replacing motion traditional detection with neural net based detection.
I previously modified motion to directly support the NCSDK and that worked really well in terms of performance and detection. Problem with my approach is that it only works with C based neural net libraries because motion is written in C. Since then, we have OpenVINO, Coral, etc, and all of them support python. Even OpenVINO C++ library is a problem for motion because motion developers are unlikely to move to C++. One way around this problem is to use [shared memory](https://www.stuffaboutcode.com/2013/08/shared-memory-c-python-ipc.html) between motion (C) and neural net module (python). This will give the best performance because we are going to be transferring raw images at high frame rate. The shared memory implements a generic interface between motion and the neural net module, i.e. motion sticks raw images in, expects detection bounding boxes back. This should allow us to replace the neural net module with anything we want.

wb666greene, 2019-05-02:
@jasaw 
I'm very familiar with shared memory, and that was to be my initial approach until I discovered MQTT.  I'm using MQTT as a form of IPC as its very efficient with minimal programming burden, and makes for an easy "split: of functions across multiple machines if one encounters perfomance issues.

I prefer the idea of splitting the AI to a different system with multiple ways of getting frames to process instead of my initial "monolithic" approach, that is the direction I'm moving for my next version.  But the beauty of MQTT is you can use it on the local host for IPC and then split the function to multiple machines if necessary to get desired performance targets.

My understanding is Motion processes "frames" and pipelines them to ffmpeg to make videos, dumping them via MQTT to the AI subsystem basically just requires a  "tee" in front of the pipeline, if no MQTT subscription for them the Broker just dumps them.  The more I learn and play with MQTT the more I appreciate the beautiful simplicity of it.

I was less than clear in my need for the motion focus.  The MobileNet-SSD actually works way better than seems reasonable considering the image is resized to 300x300 for processing) on everything from QIF to full HD, but I've been trying some 4K cameras and there its not so good at detecting people.

My bias is against false positives, if the frame rate is decent false negatives will become true positives if the person lingers in the field of view.  I want "drop what you are doing and respond" alerts which requires near zero false positive rate,

Did I post the link to this clip here?  Its basically my motivation, its a clip lifted from a local TV news website and ran through a very early version of my AI code (Using OpenCV to decode the mp4 video file): [https://youtu.be/nnSTCG6u3ck](url)  Basically when my system is armed in "at home" mode I get a loud audio alert on the first "green box" from the AI, If in "not home" mode it sends SMS texts and Email images of the detection frames in addition to the audio alerts (hope is the voice alert could scare them away).

I agree that if MotioneyeOS can send snapshots (and/or) videos via MQTT as an option instead of using ftp that would be a great place to start.  Motioneye already handles the archiving, reviewing, etc. functions very nicely and an  AI addon would really push the false alert rate towards zero.  MotioneyeOS wouldn't need to implement the Broker, I'd envision the Broker to be either a standalone (or cloudbased middleman HiveMQ, IBM, etc. sells MQTT broker services) as the AI machine would generally be more powerful -- basically a decent i3 (4025U or better) is about as fast as the NCS with CPU AI and the OpenCV DNN module, a decent i5 (4200U or better) is basically as fast as the NCS2.  Refurb systems with these specs are easy to find in the $150-250 price range.

I could have had a video file "front end" option for the code on my GitHub (what I used for the video clip) but I decided not to, as video file formats are a potential nightmare to deal with.  Decoding rtsp streams (basically mp4 h264/5) Works well on i5 or better machines, not so great on the Pi3B+ but usable for a couple of HD streams if the frame rate is reduced to ~5 fps for each stream.

I'm definitely interested in contributing to get MotioneyeOS sending frames via MQTT as an option along side ftp.  I just need some guidance as to where to start.

MrJBSwe, 2019-05-04:
https://arxiv.org/pdf/1904.08653.pdf
[image]

wb666greene, 2019-05-05:
@MrJBSwe 
This is most interesting, but folks using Raspberry Pi and MotioneyeOS are not protecting Ft Knox!

Just two nights ago scumbags stole the tires and rims from two large Chevy/GMC pickup trucks parked in their driveways just down the street from me.  These mopes just ain't that smart.  

If my neighbors had my AI enhanced security system looking at their driveway, the audio alerts would have quickly introduced the perps to Mr. 12Gauge before they got the first wheel off!  (this is Texas, being a thief in the night is extremely dangerous work!)

I have downloaded the pdf file, thanks for passing it on.

scottlamb, 2019-06-04:
[I hope no one minds my popping into this thread. I'm not a motionEyeOS user; I hadn't even heard about it until this thread popped up in search results. But I'd like to do similar things with my own NVR software and am really interested in what folks are saying here.]

I'd love to have high-quality motion detection on, say, 8 video streams at 10 fps each, with cheap hardware. The obvious approach is to have each camera responsible for its own motion detection, but the cheap cameras' built-in motion detection is crappy, manufacturers are apparently hostile to firmware modification / open source, and I think said cameras don't have NPUs (yet at least), so that doesn't seem feasible. I'd be thrilled if I could do everything with one of these sticks plugged into the NVR machine. There also seem to be some SBCs coming out with NPUs built in (eg the upcoming [Khadas VIM3](https://www.khadas.com/vim)), which might be even cheaper (given that you're buying an SBC anyway for the NVR itself).

The comments above seem to be describing object detection networks (like YOLO), which apparently isn't as fast as I'd like on current hardware. (Thanks for posting numbers!) Hardware will get better over time, but I'd also guess that pure motion detection (aka change detection, background segmentation, background subtraction, foreground segmentation; they seem to be near-synonyms) would be significantly faster.

Has anyone tried using a change detection network instead? (There are some listed at [changedetection.net](http://changedetection.net]).) I'd be very interested to hear what performance numbers you get and if you were happy with the results. I think these networks are significantly better (in terms of false positives and false negatives) than classical approaches like what motion does today.

I don't (yet) have any of this hardware to see for myself. I'll probably buy one to experiment with after I check off some other things on my software's todo list.

For the change detection approach, I don't think dropping frame rate too much would be feasible. changedetection.net has a "Low framerate" category, and the results are significantly worse than the "Baseline" category. I think low resolution is less problematic. (The current top-ranked method, FgSegNet_v2, was apparently run at 320x240. In terms of bandwidth, 8 uncompressed 10 fps grayscale streams at that resolution total under 50 Mbps.)

jasaw, 2019-06-04:
@scottlamb Check out the performance table here: https://github.com/weiliu89/caffe/tree/ssd
Single Shot Multibox Detector (SSD) is faster and more accurate than YOLO. I've been playing with MobileNet SSD (300 x 300 pixel) and it has the best balance between speed and accuracy, certainly better than tiny-YOLO in every way when it comes to security camera application. I used Tiny-YOLO because the full YOLO neural net model is either too big to run on NPU, or way too slow to be practical.

I'm not sure what you mean about the dropping frame rate feasibility. For real world security camera application, I have cameras running at high frame rate, but running inference at a much lower frame rate (5 fps), and it works great at detecting people. If someone can walk in and out of a camera view within 200ms (5fps), then this person won't be detected. This real world scenario is very different from the "change detection" tests. Anyway, 5 fps is limited by NPU hardware capability. When NPU hardware gets more powerful, this 5 fps won't be a problem anymore.

The low resolution of these detection neural nets is not a problem at all. Lower resolution buys you a lot of speed at the cost of some accuracy. Checkout the comparison between SSD 300x300 pixels vs 512x512 pixels. The higher resolution buys you 2% increase in VOC2007 test score at the expense of halving the frame rate.

wb666greene, 2019-06-04:
@jasaw 

> Anyway, 5 fps is limited by NPU hardware capability. When NPU hardware gets more powerful, this 5 fps won't be a problem anymore.

We may be getting there pretty quickly.  My initial tests with an Nvidia Jetson Nano (~$100) and a Coral USB TPU (~$75) I'm getting ~40 fps inference using MobileNetSSD, 

IMHO "change detection" is worthless, I don't care about changes in the image, I care about the presence of people when and where they shouldn't be.  You want object detection, particularly "people detection" not  change detection. 

I'm of the opinion that ~2 fps per camera is fine for the "sentry duty" of AI person detection, refer to the video I linked in my comment on May 2

I run my 16 channel Lorex DVR at 5 fps per camera for 24/7 recording and the AI runs about 1 fps per camera.  I can improve the AI frame rate by processing the 5 fps rtsp streams but that adds 2-3 seconds to the detection latency.  I'm actually more interested in reducing detection latency that improving frame rate.

scottlamb, 2019-06-04:
@jasaw Yeah, I saw your comments above about MobileNet SSD. As I understand it, that does object detection—tells where in the image each instance of N classes (humans, cars, monkeys, etc.) of object is.

And of course on the other hand motion's current approach is change detection based on relatively simple classical algorithms: basically "x pixels changed" (by more than a certain luminance, after applying erode/dilate, correcting for "light switch", etc), which has trouble filtering out "boring" changes (shadows, insects, spider webs, trees blowing in the wind, rain, small mammals, etc). I have this same trouble with cheap proprietary cameras' built-in motion detection.

changedetection.net has a variety of other methods (both more sophisticated classical approaches and neural net-based ones) that I believe are supposed to be something between. Not necessarily able to label human vs car but able to distinguish a large object crossing the frame from a smaller one or uncoordinated movement. I think if these live up to their promise, they'd be sufficient for my needs and perhaps those of other people here. And I'm wondering if they're significantly faster than objection detection—enough so that one accelerator stick could handle many video streams rather than just one. A caveat I'm noting is that they may need a higher framerate than the object detection approach you're using, so they'd have to be considerably faster on a frame-by-frame basis to have the benefit I hope.

My question is: has anyone evaluated the changedetection.net methods? It sounds like you and @wb666greene are skeptical of them. Fair enough. Is that based on intuition or experience? I'd be very interested to know if anyone has tried them and if so what they found.

@wb666greene 40 fps of object detection sounds exciting! At 5 fps per stream, that's the 8 streams I was hoping for. (I also hear you suggest even 2 fps is sufficient.) This may make my question above moot, but I'm still curious.

I agree with both of you that the frame rate (and resolution) you do detection on can be considerably less than what you retain for viewing. Currently I save two H.264 streams per camera: 1080p/30fps (main) and 704x480/10fps (sub). I'd definitely do the H.264 decoding and detection only on the sub stream and could downscale it to 300x300 or whatever on the CPU before sending to the NPU, could only send every other frame if doing object detection, etc.

wb666greene, 2019-06-04:
"My question is: has anyone evaluated the changedetection.net methods? It
sounds like you and @wb666greene <https://github.com/wb666greene> are
skeptical of them"

Circa 2013 I put a lot of effort trying to improve the false alert rate of
my Lorex HD security DVR system using a combination of PIR motion sensors
and "traditional" image processing with OpenCV,  none were close to good
enough.  AI became accessible circa 2015 when Google open sourced
TensorFlow.  My first tests with AI, while far too slow for my needs at the
time, the results of the detections on real security camera images from my
system told me this was the future.  It became "practical" with the release
of the Intel Movidius NCS circa 2017.  Improvement has been dramatic since
then.  I also don't care about detecting anything other than people,
MobileNetSSD is laughable if you draw boxes for all its 21 "objects" but it
seems to be the best so far for detecting people with resource constrained
AI.

Its still not perfect.  My main interest in "doubling" the frame rate is
for an attempt to further reduce the false detection rate.  My idea is to
us the initial detection bounding box to "zoom in" on the image and rerun
the detection.  I'm collecting test images and working on batch processing
code to test the idea,  Attached is a rare false detection, apparently a
bug flying near the camera lining up with background "clutter" and the
harsh lighting of the setting sun.  Only got this once, the system has been
running 24/7 for almost a year now, but I've other "bug" events.  Some
"static location" false detections from  plants are being filtered out by
the location and size of the bounding box.  But the insects issue needs a
dynamic solution.

scottlamb, 2019-06-04:
I agree neural network-based methods are the way to go. Your report of 40 fps and very few false positives just convinced me to buy a Coral USB accelerator—it should be here Thursday. I may experiment a bit with change detection neural networks if no one else has, but it's great to know the object detection path is working out well.

I also wonder if an object detection model trained to detect only people (and not MobileNet SSD's 20 other classes) would be significantly faster. This is all a little outside my comfort zone but I'd like to develop an understanding of it sooner or later...

wb666greene, 2019-06-04:
I need to point out the 40 fps is running on an i3-4200 or an Nvidia Tegra
Jetson Nano.  I'm getting about 20 fps on an Odroid XU-4 and only about 6.5
fps on a Pi3B+ (lack of USB3 hurts much more for the Coral than it does for
the NCS)

scottlamb, 2019-06-04:
Thanks for pointing that out; my main installation is on a ODROID-XU4 now. Do you happen to know what the bottleneck is there, given that the ODROID-XU4 has USB3? I can try to figure it out if not.

wb666greene, 2019-06-04:
I can't say for sure at this point, but the Jetson Nano has a Cuda capable
GPU so some of the OpenCV image conversion stuff may be accelerated, and it
has a lot more RAM.

bsdwatch, 2019-06-10:
I already have a Nvidia NANO with motion / motioneye onboard ... Ive gone a different route here then this project as generating images with yocto was far easier to accomplish, hence there is no raspbian or othher for the Nano... yocto it is.... that being said, i have an image with the software onboard, now im running down cuda / opencv into the build. Stay tuned....

wb666greene, 2019-06-10:
I'm certainly interested in your results with YOLO.

YOLO or tinyYOLO?

YOLO was great in my early tests but at 14 second per inference on my i7
desktop I gave up on it immediately. My tests with tinyYOLO showed that
while it was closer to being fast enough, it performed terrible on "people
detection" with test images from real security cameras.

MrJBSwe, 2019-06-13:
24TOPS 0.7W
https://www.youtube.com/watch?v=yQvVqaVZUQ4

https://www.gyrfalcontech.ai/solutions/2803s/

wb666greene, 2019-06-13:
Its nice to have options, and at $69 "pre-order" price its competitive, but
its all going to depend on the tools and systems supported.
I'm working on adding the Coral TPU to my AI code, pretty impressive in
simple tests.  The learning curve on these things is not insignificant.

[52 more replies]