motioneye-project/motioneyeos #1507: 5ghz support on pi 3b+

shadowfxd, 2018-05-02:
i keep trying to get 5ghz working with the latest pre release on a pi 3b+ and its not working. its only working on 2.4ghz and ethernet on pi3b+.

ccrisan, 2018-05-03:
I can confirm this is a valid, reproducible issue with 20180401 on RPi 3B+.

shadowfxd, 2018-05-03:
Will you be working on a fix, so we can use the 5Ghz on Pi 3B+? If so, any ETA?

ccrisan, 2018-05-03:
@shadowfxd of course I will :) I'll have to dig deeper into this stuff as I cannot possibly tell by now what's wrong with it. And thus I cannot come up with an ETA, but I can tell for sure that it's not going to happen during this weekend.

Also my build machine is still broken so I need to do all the builds on my laptop, which is very noisy, heaty and time-consuming. But I'll get to the bottom of it eventually and there will obviously be a fix for this.

shadowfxd, 2018-05-03:
Thanks @ccrisan , appreciate all your work!

smithatlanta, 2018-05-17:
My RPi 3B+(with 20180401) is connecting via 5ghz to my network but it is so slow it's unreachable most of the time.  I think it may be an issue as much with my mesh network as the pi, though.