motioneye-project/motioneyeos #1508: Motioneye not starting ubuntu 18.04

wolfi74, 2018-05-04:
Hi,
since update to ubuntu 18.04 motioneye wont start working.
The service is not started.
Please help.
Thanks

chrisspen, 2018-05-12:
What do the logs say? What error message do you get if you try to start it manually. No one can help you with an error report that simply says "Motioneye not starting".

wolfi74, 2018-05-28:
@abscode0 thank you very much. Works for me too

jasaw, 2018-06-06:
Looks like problem has been resolved. Closing issue.

closed 2018-06-06 by jasaw

wandering213, 2018-06-20:
This issue is caused by old cache information left over from the old version of 
Ubuntu. Run rm -rf ~/.cache/pip and reinstall pycurl. 

https://stackoverflow.com/questions/50683895/importing-pycurl-with-libcurl4-raises-importerror