motioneye-project/motioneyeos #1512: The best small board video cam program in the world, but no sound.....tragic

keelobight, 2018-05-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (pi zero w, pi3b).

### Camera

I am using the following type of camera: many

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (wifi).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

jasaw, 2018-05-10:
See https://github.com/ccrisan/motioneyeos/issues/105

closed 2018-05-10 by ccrisan

keelobight, 2018-05-10:
Oh thanks.  I'm actually trying to make a baby monitor too that my wife can just view on her phone around the house.  Absolutely love your OS, best I've ever seen I use it on all my cams in my business.  I've been up all night trying to find a solution I'm surprised how hard it is, there's just not much out there that works well.  And I'm no programmer so I have to find something intermediate level to implement.  Well thanks for writing back!

joshikoff, 2018-05-13:
I am more than happy having finally a good solution for cam controlls. Thank you for your efforts.

The more I think, the more I wonder why I have spent so much money on commercial / semi comercial solutions which finally did do a bad job.

I have 6 cam´s hooked up thru a RPi 3 and it works fantastic! Thank you so much.