motioneye-project/motioneyeos #1515: streaming URL with user/password

hans720, 2018-05-12:
I am trying to connect to the motioneyeos image stream and I have authentication turned on. I am trying to embed the username and password in the URL/
I have tried the following and neither work

http://username:password@ip:8081
http://IP:8081?user=username&pwd=password
http://IP:8081?user=username&password=password

Is there another pattern I should try or is this not possible.

thanks a lot ..

tigernero79, 2018-05-13:
Test
http://IP:PORT/videostream.cgi?user=admin&pwd=pasword

Akeid, 2018-05-14:
I believe he is trying to hook into the stream generated by a motioneye camera. I'm not sure how to do this with authentication enabled.

ccrisan, 2018-05-14:
@Akeid the syntax `http://username:password@host" would simply instruct the HTTP *client* to use username and password as credentials, if needed. The URL does not get to the server including credentials. Long story short, the syntax is not specific to motionEye and, in general, has nothing to do with the HTTP server.

hb-echo, 2018-06-07:
hey,
I am trying to connect to the motioneyeos image stream with chromium on my 2nd Pi and I have authentication turned on. I am trying to embed the username and password in the URL like this: 
http://username:password@host"
The authentication is working but the layout is very starnge...  it only show´s a white screen with all preferences and settings... all boxes are empty and no stream.
(trying to upload a screenshot but it fails 👎)

any idea?
thanks a lot..