motioneye-project/motioneyeos #1519: overclocking pi zero w

mauzzz, 2018-05-21:
I have a pi zero w with pi camera module v2.1 and the latest release for the zero, motioneyeos-raspberrypi-20180224.img.gz. There is no overclocking setting in preferences for the pi zero. Am I using the wrong release?

jasaw, 2018-05-22:
Make sure you log in as admin. Overclock settings should be under "Expert" options. You may have to enable "Advanced options".
If that doesn't solve your problem, maybe you could attach a screenshot?

xoma262, 2018-06-04:
@jasaw I have the same isssue on Pi 3B
I attached the screenshot. Advanced checked in, logged in as admin, - nothing under expert settings. 
[image]

jasaw, 2018-06-05:
@xoma262 That is very weird. I noticed you have system monitoring turned on. Can you please attach your log files here? Can you also attach your /boot/config.txt file?

xoma262, 2018-06-05:
@jasaw I attached logs and config.txt file. Let me know if you need anything. 
Weird thing is that I have an overclocking feature on my RPi Zero W. 

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/2070789/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/2070790/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/2070791/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2070792/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/2070793/motion.log)
[config.txt](https://github.com/ccrisan/motioneyeos/files/2070804/config.txt)

iamtheyammer, 2018-12-18:
For what it's worth I'm still having this issue. motionEyeOS 20180628, Pi 3b+.

xoma262, 2018-12-18:
@iamtheyammer have you tried the latest build?

iamtheyammer, 2018-12-19:
Not yet, I’ll try toda

iamtheyammer, 2018-12-19:
Just tried it, same issue.

Versions:
motionEye Version | 0.39.3
Motion Version | 4.1.1
OS Version | motionEyeOS 20181209

ccrisan, 2018-12-26:
The Raspberry 3 variant does not support overclocking.

closed 2018-12-26 by ccrisan